mavlink/qgroundcontrol #6083: Similar issue with #5816

Jdiesel666, 2018-02-01:
Firmware halted at "programming new version" forever when trying to update on Ubuntu 16.04.
I'm using PixHawk http://ardupilot.org/copter/docs/common-pixhawk-overview.html
The QGround version is 3.2.7 right from the website https://docs.qgroundcontrol.com/en/getting_started/download_and_install.html
The version I had before doesn't have the rover airframe on it. 
I just need qgroundcontrol for the calibration to be able to use it with VINS-Mono.
I'm not sure if I have to install mavros in order to get it to work.

DonLakeFlyer, 2018-02-01:
Use a new daily build.

closed 2018-02-01 by DonLakeFlyer

Jdiesel666, 2018-02-01:
@DonLakeFlyer What is the optimal choice of airframe for Ground Vehicle? The daily build doesn't have a Rover and the compass setup kept failing.

DonLakeFlyer, 2018-02-01:
You'll have to ask firmware folks about that. I assume you are using PX4. Not sure if they support Rover.

hamishwillee, 2018-02-01:
PX4 only has basic Rover support, but the build and link to frames is here: https://docs.px4.io/en/frames_rover/

DonLakeFlyer, 2018-02-01:
But that is not in the airframe metadata in QGC hence no Rover shows up. @LorenzMeier Is the airframe meta data out of whack?