mavlink/qgroundcontrol #6095: /dev/tty.SLAB_USBtoUART Open failed. Resource busy

RonildoBragaJunior, 2018-02-02:
Connection between the QGroundControl and Pixracer 1.4 through UART(FPV Radio Telemetry Air Module) is not working.

Discussion:
http://discuss.px4.io/t/how-to-connect-qgroundcontrol-to-pixracer-1-4-through-uart-radio/5597

Hardware:
Pixracer 1.4 

Software:
MacOs High Sierra Version 10.13.3
Firmware v1.7.3 Stable Release
QGroundControl Daily Build

Driver was obtained from here:
https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers

DonLakeFlyer, 2018-02-02:
Can you pull across the log output to here as well?

RonildoBragaJunior, 2018-02-02:
Thanks... please have a look on the log below.

[log.txt](https://github.com/mavlink/qgroundcontrol/files/1691057/log.txt)

DonLakeFlyer, 2018-02-05:
Looks like more of a driver problem than a QGC problem. @dogmaphobic Do you know anything about these sorts ofr boards?

dogmaphobic, 2018-02-05:
SiLabs? That's different. Those are always nasty to get working as they have lousy drivers. See here:

https://www.silabs.com/community/interface/forum.topic.html/cp120x_vcp_driverno-G1Bu

I had the same issue with something completely different, not related to QGC but using one of their USB/UART chips.

RonildoBragaJunior, 2018-02-07:
It is solved. I believe it was a problem with the driver which I fixed. Thank you.

closed 2018-02-07 by RonildoBragaJunior