mavlink/qgroundcontrol #6099: RC calibration process "hangs" at the end of the process

jstrebel, 2018-02-02:
Hi,
I am trying to calibrate my RC. I can do all the steps until to the Point “move the Pitch stick down and hold it there” It seems to me the process does not finish. (only the cancel button is solid) 
Using QGC 3.2.7 stable on Ubuntu Linux, RC is mode 1. Pixhawk SW 1.7.3 I have done this process sucessfully some months ago using a older version on QGC and pixhawk

Jakob

jstrebel, 2018-02-03:
Same happens doing it with OSX. Jakob

jstrebel, 2018-02-03:
Solved: It seems the time window to move pitch up and down plus moving throttle down is somehow critical (respective small).

DonLakeFlyer, 2018-02-05:
> It seems the time window to move pitch up and down plus moving throttle down is somehow critical (respective small).

Can yoy explain in detail what you mean by that? I don't understand.

jstrebel, 2018-02-07:
Don, pls give me some time. Need to get the pixhawk back in my hands and do the steps again and describe the last 3 steps. Mainly it is the allowed time to move the stick in the desired position.

DonLakeFlyer, 2018-02-07:
> Mainly it is the allowed time to move the stick in the desired position.

That's the part I don't understand. There is no "allowed" time, the allowed time is forever.

jstrebel, 2018-02-13:
Don,
Attached is the description of the problem described as good as I can. In my setup it's reproducible. I hope my graphic helps you. If I am not fast enough in step 8, the setup jumps immediatly to step 9. Then exit is only possible by clicking the cancel button.
[QGC_RC_CAL.pdf](https://github.com/mavlink/qgroundcontrol/files/1720040/QGC_RC_CAL.pdf)
Jakob

DonLakeFlyer, 2018-02-13:
Now I understand. The image for roll right for Mode 1 is incorrect. It shows the throttle as being centered.. The throttle should stay down while you do this. That is confusing the system. Can you try calibration with keeping the throttle down at this step and see if it works. After you complete the steps for throttle up/down the rest of the calibration should be done with throttle all the way down.

jstrebel, 2018-02-14:
Don, if I follow the steps marked in yellow then it works.
[QGC_RC_CAL.pdf](https://github.com/mavlink/qgroundcontrol/files/1722821/QGC_RC_CAL.pdf)

jstrebel, 2018-02-14:
Don, I am willing do some testing in case you are going to fix this. 
(Platforms available: OSX or Ubuntu Linux) Jakob

closed 2018-03-22 by DonLakeFlyer

ashirsat, 2020-08-13:
Hi @DonLakeFlyer I am getting the same issue the process freezes on after calibration with only cancel button activated. I am running the latest version of QGroundControl on Ubuntu 18.04.

Please let me know if you have any suggestions?

Thanks,
Aniket

DonLakeFlyer, 2020-08-14:
"Latest version" meaning what? Stable or Daily Build? Take a screen shot of the version info which is shown at the bottom of the General Settings page and include here.