mavlink/qgroundcontrol #6101: Multiple laptop connections to QGC

RohanKurane, 2018-02-03:
This system is for posting bugs or feature requests ONLY. For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

----
When posting bug reports, include the following informaiton:
- Ground station operating system.
- QGroundControl version and build (daily, stable, self-built from source, etc.)
- Autopilot board: Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.
- Autopilot firmware (e.g. PX4, ArduPilot, custom) and version.
- Exact steps to reproduce the problem. Starting from booting QGroundControl to when the problem occurs.

**All this template text should be deleted before creating your issue.**

RohanKurane, 2018-02-03:
Hi,
Can i have multiple laptops connect to the Intel Aero RTF drone via QGC ? When i tried to connect 2 laptops via QGC to the same Intel RTF drone, the second one could not connect. 

Thanks

DonLakeFlyer, 2018-02-03:
You'll have to ask the intel folks about that.

closed 2018-02-03 by DonLakeFlyer