mavlink/qgroundcontrol #6102: How to add RTSP Video Stream to the QGC Android version and why the Android version has not RTSP Video Stream

120639850, 2018-02-03:
This system is for posting bugs or feature requests ONLY. For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

----
When posting bug reports, include the following informaiton:
- Ground station operating system.
- QGroundControl version and build (daily, stable, self-built from source, etc.)
- Autopilot board: Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.
- Autopilot firmware (e.g. PX4, ArduPilot, custom) and version.
- Exact steps to reproduce the problem. Starting from booting QGroundControl to when the problem occurs.

**All this template text should be deleted before creating your issue.**

120639850, 2018-02-03:
i am a QGC developer,now i have a network cam.i wannt to connect the network cam to the QGC Android version.but i found out that QGC Android has not RTSP Video Stream.i wannt to know why and how add code to make the Android can accpet RTSP Video Stream.

120639850, 2018-02-03:
By the way,my Andorid System has Ethernet Driver and i can accept the RTSP Video Stream in the Browser.thank you

DonLakeFlyer, 2018-02-03:
> This system is for posting bugs or feature requests ONLY.

You should ask usage question either on Gitter or the various QGC forums.

closed 2018-02-03 by DonLakeFlyer