mavlink/qgroundcontrol #6109: Feature request - Add optional, configurable, on-screen buttons for aux servo control

fredrikfalkman, 2018-02-05:
It would be useful to be able to choose to show buttons or sliders that could be mapped to aux servos.
It would be preferable if you could choose the name/description of the buttons.

This would be as an alternative to use buttons and dials on an RC transmitter. The function would be most useful for those who fly without RC.

I guess this is similar to the current "Trigger Camera" in the main instrument panel. A dedicated instrument panel screen might be a good place for these controls to live.

My immediate use for this function would be for a plane to control a remote launcher with commands like "Prepare Launch", "Cancel Launch", "Launch", "Turn Launcher Right", "Turn Launcher Left".

Other uses might be for flaps, retractable wheels, pitot heaters, etc.

GlidLov, 2018-02-06:
That's a great idea. Can be used also for gimbal control since ardupilot fw has mavlink messages to do that (trigger, do_digicam_control, etc), so a virtual dashboard could be configured with preferred commands (pilot: has main command, camera operator: has gimbal control, trigger, zoom, etc)

DonLakeFlyer, 2018-02-06:
@GlidLov You are talking about something different. This is for servo commands only. Gimbal control is already being working on separately and will be coming soon in 3.4.

GlidLov, 2018-05-08:
Well, glad to read that, but i'm not referring only to gimbal rather Zoom, Trigger and maybe a configurable way to add commands in "button" style (not proportional input needed)

DonLakeFlyer, 2018-05-08:
I believe @dogmaphobic  may have alreaded gimbal commands to the instrument panel camera page. Camera trigger was already there.

manavgandhi17, 2022-02-25:
Has there been any update on this matter. Looking for something similar to what the "Servos" tab is in Mission Planner.

manavgandhi17, 2022-09-12:
Pinging to see if any work has been done on integrating this feature.

zdanek, 2022-09-12:
I have almost built something that might be accurate. A mapping of a buttons to PWM values - low and high. With latch.
 If you press a button, RC override is sent via mavlink - a defined high value. If button is released  a defined low value is sent.
Latch means rotating high and low at every button press.

I'd love to show you a screen with this but I left  joystick at the office. It required joystick to enable button assigment panel.
But on Thursday I will post a screen to show you what I mean.

GlidLov, 2022-09-13:
zdanek, it seems a good point to start with experimenting. I'm definitely interested on integrating this feature

closed 2026-10-04 by github-actions[bot]