mavlink/qgroundcontrol #6114: RAKWireless RTSP video works in IOS but don't work in Android (green screen, lags)

GlidLov, 2018-02-06:
---
- Ground station operating system: Android
- QGroundControl version: 3.2.7
- Autopilot board: any
- Autopilot firmware: any
- Exact steps to reproduce the problem:

trying to get a proper video stream from the RAKWireless Flylink set (it's a digital, low latency module which accept HDMI as input and streams RTSP over a wifi network).

steps to make it work:
1) power up everything (tx, rx, camera)
2) connect phone/tablet/pc to newly created wifi network
3) open (or configure QGC) a rtsp link as follows:  rtsp://admin:admin@192.168.100.100:554/cam1/h264

followed those steps into multiple devices:
IOS (iphone 6) and QGC: Works perfect, video appears into window and good latency (~100ms or so)
PC (win10): works ok

android (my development and use platform) and QGC: video appears, but totally greened, heavily pixelized and lagged by 2-3 sec

android with RAKwireless test app: works ok, latency and color ok, same as iphone.

Rakwireless team has been contacted already and it's very cooperative, but at this point i think there's a problem with QGC and it's inner way to manage RTPS, or device drivers/codecs etc.

Since i have all the hardware needed i can do testing if a workaround is released.

Good work!

dogmaphobic, 2018-02-06:
The RTSP code running in QGC is identical in all platforms. What differs is hardware decoding, which is defined by the platform itself. Have you tried this on a different Android device/brand? Beyond that, it will be hard to identify the problem, which so far seems unique to this RAKWireless unit.

For curiosity's sake, can you run ffprobe and paste the results here?

```
ffprobe rtsp://admin:admin@192.168.100.100:554/cam1/h264
```

GlidLov, 2018-02-12:
Hi,

i tried with different Android platforms and they all have same problem.
Can you tell me exaclty how to run ffprobe under Android?

dogmaphobic, 2018-02-12:
You don't need to run it under Android. I'm curious about the stream, not Android. You can run ffprobe from any platform.

GlidLov, 2018-02-16:
Dogmaphobic, i've done some search but could not find how to run ffprobe on my android, at least in a "easy way". Can you tell me some instruction about it?

GlidLov, 2018-02-27:
here you are:
-----------------
C:\>ffprobe rtsp://admin:admin@192.168.100.100:554/cam1/h264
ffprobe version 3.4.1 Copyright (c) 2007-2017 the FFmpeg developers
  built with gcc 7.2.0 (GCC)
  configuration: --enable-gpl --enable-version3 --enable-sdl2 --enable-bzlib --enable-fontconfig --enable-gnutls --enable-iconv --enable-libass --enable-libbluray --enable-libfreetype --enable-libmp3lame --enable-libopenjpeg --enable-libopus --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libtheora --enable-libtwolame --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxml2 --enable-libzimg --enable-lzma --enable-zlib --enable-gmp --enable-libvidstab --enable-libvorbis --enable-cuda --enable-cuvid --enable-d3d11va --enable-nvenc --enable-dxva2 --enable-avisynth --enable-libmfx
  libavutil      55. 78.100 / 55. 78.100
  libavcodec     57.107.100 / 57.107.100
  libavformat    57. 83.100 / 57. 83.100
  libavdevice    57. 10.100 / 57. 10.100
  libavfilter     6.107.100 /  6.107.100
  libswscale      4.  8.100 /  4.  8.100
  libswresample   2.  9.100 /  2.  9.100
  libpostproc    54.  7.100 / 54.  7.100
[udp @ 073668c0] 'circular_buffer_size' option was set but it is not supported on this build (pthread support is required)
[udp @ 07376b00] 'circular_buffer_size' option was set but it is not supported on this build (pthread support is required)
[udp @ 073874a0] 'circular_buffer_size' option was set but it is not supported on this build (pthread support is required)
[udp @ 073976e0] 'circular_buffer_size' option was set but it is not supported on this build (pthread support is required)
Input #0, rtsp, from 'rtsp://admin:admin@192.168.100.100:554/cam1/h264':
  Metadata:
    title           : h264
  Duration: N/A, start: 0.000000, bitrate: N/A
    Stream #0:0: Audio: aac (LC), 48000 Hz, stereo, fltp
    Stream #0:1: Video: h264 (High), yuvj420p(pc, bt709, progressive), 1280x720, 29.97 tbr, 90k tbn, 180k tbc

dogmaphobic, 2018-02-27:
>Stream #0:1: Video: h264 (High), yuvj420p(pc, bt709, progressive), 1280x720, 29.97 tbr, 90k tbn, 180k tbc

The only thing that stands out is the fact it is using a _high profile_, which isn't exactly suitable for streaming. It should be using an _Extended Profile_ instead. With that said, there isn't much I can do without an actual camera to test with.

gevab, 2018-07-20:
Hi,

Same issue here with QGroundControl v3.2.5 on Mac OS X 10.11.6 !

Is there anything we can do in order to move forward and analyze this issue ?

GlidLov, 2019-03-09:
only workaround i found is to use the CVBS version (analog input) instead of the HDMI input of the RAK transmitter. This way, it works on all platforms ok. Video quality is pretty bad and blocky, thou.

dogmaphobic, 2019-03-09:
Are there any streaming settings? Can you change the data rate, video resolution, encoding, etc?

GlidLov, 2019-03-14:
There's an "app" which Connect to the thing which has some settings. I Will post a screenshot

closed 2026-10-04 by github-actions[bot]