mavlink/qgroundcontrol #6125: ARM PX4 by the RC in mode 1

jstrebel, 2018-02-07:
As far I understand arming can be done by the throttle stick. 
- Put the stick back to "zero" throttle" and then move it to the right position.
In Mode 1 I discover only the following stick procedure works:
Put the throttle stick (stick on the right) to "zero"
Move rudder Stick (stick on the left) to the right. -> PX4 is armed
Move rudder Stick to the left. -> PX4 is disarmed
Jakob

DonLakeFlyer, 2018-02-07:
And the problem is?

jstrebel, 2018-02-08:
My Problem is solved. My intention was, helping other users who fly mode 1 not falling in the same trap.
I found just by coincidence arming in mode 1 via the RC stick has to be done using the rudder stick on the left. 
My suggestion is, to mention this in the documentation.

closed 2018-02-08 by DonLakeFlyer