mavlink/qgroundcontrol #6126: fixed HUD and value widget in 3.3

taileron, 2018-02-08:
pls bring the nice slim rectangular HUD / value widget from 3.2.7 back again. 
Displaying compass around HUD was an absolutely great throw.
The new one is too wide, ugly and always covers too much map and video content while hud display is too small. 
Or enable aligning compass below HUD to reduce with.

DonLakeFlyer, 2018-02-08:
Some people complain not wide enough, some people complain too wide.

Antiheavy, 2018-02-10:
It took me a while to warm up to the new side-by-side HUD widget.  I now prefer the new one to the rectangle one and to the older "vertical pill" too.

I've learned proper compass calibration is a critical flight safety issue and the small heading pip around the rectangle was too small for me and others to pay attention to during pre-flight.  A dedicated compass display gives proper importance to this user required sensor check during preflight.

I do think the "values" box is dis-jarring though.  Ideally it could be hidden for most users if altitude and speed where always shown elsewhere.

@DonLakeFlyer , I will humbly suggest an improvement to the "Values" box: consider small round corners for consistency with the Guided toolbar:
[image]

taileron, 2018-02-12:
... sorry for complaining this way, but the main issue is: the value widget takes a way bigger amount of space in relation to the displayed content. The best could be to display them in a  kind of overlay or transparency to avoid hiding all that behind. And hud and compass scalable with an option to align them different like the video/map overlay which I think it´s absolutely great done.

Antiheavy, 2018-02-12:
> hud and compass scalable with an option to align them different like the video/map overlay which I think it´s absolutely great done

neat idea.

I've also though it might be neat if clicking on the artificial horizon brought up a selector menu to choose either of the old styles... of course then the developers would need to continue to support the old styles which they may or maynot want to do...

I'm going to move my corner comment to a separate issue.

mrshannon, 2018-04-03:
Regardless of styles or anything else.  There needs to be a way to configure the size of the hud.  It is way too small now.

sparky002, 2018-04-20:
Speaking of HUD. Why hasn't a HUD similar to Mission Planners been integrated into QGroundControl yet? Nevertheless, a combined HUD similar to ANY modern aircraft in existence would be nice to have.

Antiheavy, 2018-04-20:
> Speaking of HUD. Why hasn't a HUD similar to Mission Planners been integrated into QGroundControl yet? Nevertheless, a combined HUD similar to any or most modern aircraft in existence would be nice to have.

This request comes up in various forms fairly often.  However, I'd argue the following:
- requests for more detailed instrument displays (6-pack clusters, mutli-function HUDs, etc) typically come from pilots and hard-core drone geeks.
- the vast majority of drone "users" (who are not pilots or hard core drone geeks) do not want all the complexity and in-fact are highly intimated by it.

In my opinion the default instrument display should be as simple, minimal, and clear as possible to match the 80% use case.  The "widget" plug in architecture makes it fairly easy for a software developer to make his/her own custom instrument display.

I'd also argue that **speed** and **altitude** should be cleanly and simply integrated into the instrument cluster (maybe just as simple, clear text) so that the big ugly white/gray box underneath can be hidden for 80% of users.

Antiheavy, 2018-04-20:
> This would also handle things like the vast disparity between a large monitor connected to a large computer and a minute screen on a mobile device.

@taileron 's suggestion of the instrument cluster being able to scale similar to the PiP video/map window would partially address this.  I realize scaling a video/map window is a totally different technical problem than scaling a graphical widget, but if the UX would have some consistency anyway.

taileron, 2019-08-15:
On some platforms, the transparency of the title bar disappears, is there a trick to increase the transparency a bit more to maintain the 16:9 ratio of the video stream?

norimboo, 2023-03-03:
and there will be great if there will be an option for full screen camera view

DonLakeFlyer, 2024-10-13:
This is all different now. The value display can be modified by the end user and you can select from different HUD visualization options: large, small and so forth

closed 2024-10-13 by DonLakeFlyer