mavlink/qgroundcontrol #6129: Mission transfer error when connected with companion computer

myeabhishek, 2018-02-09:
Hi I am using pixhawk 2 with v1.7.2 firmware. It is connected to a companion computer via TELEM2 port (921600, 8N1). The companion computer is running ROS and MAVROS.

The issue that I am facing currently is that if I connect a companion computer and try uploading a mission, then I get the following error in the QGC:

"Vehicle sent unexpected MISSION_ACK message, error: Mission accepted(MAV_MISSION_ACCEPTED)"

As soon as I switch off my companion computer or remove TELEM2, then it works fine.
Please help!

DonLakeFlyer, 2018-02-09:
What version of QGC are you using?

myeabhishek, 2018-02-09:
I am using 3.2 version. Could tell me what could be the reason behind this? and if its a bug in the QGC?

DonLakeFlyer, 2018-02-09:
Switch to new Stable. 3.3 is out now, but not yet announced.

myeabhishek, 2018-02-09:
Is this issue resolved in the latest version already?

DonLakeFlyer, 2018-02-09:
Lots of fixes in mission protocol. Also less complaints about unexpected things. First step see if QGC can handle this. Second step, figure out why turning companion computer caused problems.

myeabhishek, 2018-02-09:
All right, Ill check.

DonLakeFlyer, 2018-04-30:
Have  you tried never versions?

closed 2026-10-04 by github-actions[bot]