mavlink/qgroundcontrol #6131: Cant change to HOLD mode in QGC

RohanKurane, 2018-02-11:
I am using QGCv3.2.7. 

I tried to change to HOLD mode but it wont. What is the MAV_CMD that is sent to change to HOLD mode ? 

Is the HOLD mode same as LOITER mode ? I want to change to HOLD mode and arm the vehicle and then takeoff to a certain altitude where it should hold till I send a RETURN_TO_LAUNCH command. 

How can I do this ?

RohanKurane, 2018-02-11:
For LOITER mode the mavlink command to send is MAV_CMD_NAV_LOITER_UNLIM. 
What is the mavlink command to send for HOLD mode ? I want to be able to takeoff in HOLD mode and then when I reach a certain altitude, set the mode to HOLD so the RTF holds at that position. It does not seem to accept my MAV_CMD_NAV_LOITER_UNLIM command. Has anything changed ? Then I plan to send the MAV_CMD_NAV_LAND command to land the RTF. 
Please let me know the mavlink command for the HOLD mode. And the documentation for it.

RohanKurane, 2018-02-11:
Moved to mavlink where I thought it would be more appropriate. Leaving this open unless anything thinks otherwise.

DonLakeFlyer, 2018-02-12:
Firmware support forms are the best place for these questions. That said if you want to see how QGC does these sorts of thing look at: https://github.com/mavlink/qgroundcontrol/blob/master/src/FirmwarePlugin/PX4/PX4FirmwarePlugin.cc

closed 2018-02-12 by DonLakeFlyer