mavlink/qgroundcontrol #6137: Fixed Wing Landing Pattern

cody303, 2018-02-13:
Using the Pixhawk and Ardupilot code base the landing pattern has an orbit to altitude then a landing point. Sometimes the autopilot recognizes the waypoint as complete on the inside of the orbit radius causing the plane to turn prematurely in closer to the landing point , making a steeper decent. (see attached picture). 

[image]
I think this could be fixed with another waypoint in between the landing and the obit radius that it would need to hit before descending. That way the orbit could be placed further back, and the distance where the decent started would be at the same altitude and distance every time.

DonLakeFlyer, 2018-02-13:
@antiheavy Have you noticed this as well?

I wonder if the real fix for this is some sort of new firmware command? To force it to do at least a partial cycle around the loiter. Not sure if adding more complexity/waypoints is the way to go here.

Antiheavy, 2018-02-14:
I have not seen this issue.  Done many hundreds of landings using the QGC Landing pattern in PX4 fixed wing firmware and it pretty much always works as intended.  It always breaks out of the loiter when both the altitude is reached and the vehicle is lined up on the final approach.  Entering/holding the loiter can get a little wonky in windy conditions sometimes, but that isn't the mission plan's fault.  Ultimately it always gets lined up before exiting the loiter, even if it has to complete another time around the loiter first.

@cody303 mentioned APM, does APM fully implement the MAV_CMD_NAV_LOITER_TO_ALT to include the "Heading Required = True" field?  This is what forces the exit criteria to also be lined up with the landing waypoint.  I couldn't find where in the APM code this is implemented, but I easily could have missed it.  Maybe someone who knows more about APM could verify if implementing this would be the real fix.

DonLakeFlyer, 2018-02-14:
@cody303 You were seeing this on PX4 as well? My thinking was similar to Todd's, that something possibly wrong with ArduPilot MAV_CMD_NAV_LOITER_TO_ALT implementation.

cody303, 2018-02-14:
Yea I don't think it supports the heading requirements.  We are not currently using PX4, but maybe it is time to switch. 

Cody

DonLakeFlyer, 2018-02-14:
It needs to support MAV_CMD_NAV_LOITER_TO_ALT.param4 = 1 which indicates a tangent cross-track exit. I had thought it supported that. But I have a feeling they may have implemented in a different way which doesn't quite end up working to same all the time. Whereas PX4 always circles the loiter and exits.

DonLakeFlyer, 2018-02-16:
So not much I can do here.

DonLakeFlyer, 2018-02-18:
@cody303 I'm going to close this. Looks like ArduPilot needs to do two possible things:
* Support tangent cross-track exit
* or if they do it appears to have bugs where in some cases the tangent exit is not in the direction of the next waypoint

closed 2018-02-18 by DonLakeFlyer

kikislater, 2018-04-05:
So this one will never work with Arduplane ?

DonLakeFlyer, 2018-04-05:
You'll have to ask the ArduPlane folks about that. I would suggest providing them with an example where it doesn't quite do the right thing and flight logs in an Issue.

magicrub, 2018-04-06:
@cody303 @DonLakeFlyer  This all definitely works in ArduPlane but there is a corner case that just got fixed here https://github.com/ArduPilot/ardupilot/issues/7934 where if you're using a LOITER_TO_ALT but performing a turn and you're already at the altitude in the first couple seconds of the loiter then the circle is not much of a circle. This was not planned to be in the current 3.8.5beta but I think it would be a good candidate to include.

DonLakeFlyer, 2018-04-06:
Ah, fantastic.