mavlink/qgroundcontrol #6139: Android: QGC likes all kinds of USB devices

AndKe, 2018-02-13:
QGC daily for Android. 
I brought this up half a year ago, now it's too annoying to be ignored.

Then, I reported in on Note4 (and Note 10.1 2014)
Now I have a new Note 8 - and same issues right after installing QGC: 

Whenever I connect an USB adapter like logic analyzer, thermal camera, Gear VR, or almost "anything" except of USB storage, Qgroundcontrol pops up asking if I want to use it for that device. 

I know that's the OS, not QGC asking, but somehow QGC "told" Android that it's willing to handle almost any device.  (I did not check if all those devices enumerate as (among other things) an serial interface)

What I know, is that Tower, Solex, and no other software for the other devices I may connect behave that way.  

I plug in a Thermal camera , and I am offered to open it's app or QGC
I plug in an logic analyzer, offered to open it's app or QGC
I plug in an 3DR radio, and QGC is the only option.

no other apps I have respond to "everything"..

DonLakeFlyer, 2018-02-14:
Yup. It's implemented like that on purpose. It was better that way than the reverse which is it doesn't recognize stuff. Serial on Android with a C++ application is a royal PITA as opposed to native Java.

AndKe, 2018-02-14:
could it maybe register it's willingness to take all those devices only when QGC is running ? it's annoying for everyday use of other devices. 
Luckily "no other" app needs to behave like that, QGC should not need to do that.

mandulaj, 2020-03-23:
Hi, is there any update to this? It was getting so annoying I had to uninstall the app. However I still want to use it. Is there no option to disable the automatic app opening? I am fine with starting the app manually when I connect telemetry to the phone.

DonLakeFlyer, 2020-03-24:
Note that I'm only talking about the automatic launch fo QGC when you plug in a USB device. There is no way to prevent the Android prompt for which app to open (with QGC listed) when you plug in a USB device.

khimaros, 2025-07-22:
there doesn't seem to be a way to make it remember "no"

DonLakeFlyer, 2025-07-27:
I'll take another look at this in 5.1. Need to poke at the Tower/Solex source to see how they do it better.

DonLakeFlyer, 2026-03-12:
@HTRamsey Can you look at this? I think usb in android is all different code now isn't it?