mavlink/qgroundcontrol #6141: QGC doesn't play well with other Follow Me sources (DroneCore)

hamishwillee, 2018-02-14:
DroneCore can act as a source for PX4 follow me mode - supplying position updates ([Guide here](https://docs.dronecore.io/v/develop/en/guide/follow_me.html)). However it can't be used with QGC because as soon as the mode changes to Follow Me QGC starts sending follow me updates - in our  [test](https://github.com/dronecore/docs/pull/94#issuecomment-365507954) QGC starts sending (Lat, Lan) = (0, 0).

I'm not sure of the right solution. Some thoughts:
- QGC should recognise that it did not initiate Follow Me mode and not send updates
- QGC should not send updates if it does not have a valid location source (partial solution)
- Other?

> Ground station operating system.
>  QGroundControl version and build (daily, stable, self-built from source, etc.)

Doesn't matter. (Tested 3.2 on Linux)

>Autopilot board: Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.

Doesn't matter. Tested on simulator.

> Autopilot firmware (e.g. PX4, ArduPilot, custom) and version.

Quad.

> Exact steps to reproduce the problem. Starting from booting QGroundControl to when the problem occurs.

We run the [Follow Me Example](https://github.com/dronecore/docs/pull/94) with QGC attached on a linux laptop and Sitl. The example is already in the develop source tree - the link is to the docs that are about to be integrated.

hamishwillee, 2018-02-15:
@julianoes FYI

julianoes, 2018-02-15:
@hamishwillee I don't know follow me, I'm afraid. I'll let @shakthi-prashanth-m respond.

hamishwillee, 2018-02-15:
@julianoes OK. It is more a QGCt really a "follow me" question -

shakthi-prashanth-m, 2018-02-15:
I think QGC can't distinguish whether Mode change was done by user or detected in heartbeat.
From https://github.com/mavlink/qgroundcontrol/blob/4e9b1de50fea7a473facf799e2b5da8f5ad05798/src/FollowMe/FollowMe.h#L25

Follow Me slot `FollowMe::followMeHandleManage` is triggered when Flight mode in QGC tool is **FollowMe**.
https://github.com/mavlink/qgroundcontrol/blob/e37b067c0c77e7993685f761d02c4c232db1b91f/src/Vehicle/Vehicle.cc#L208-L209

Then it starts sending Motion report to vehicle at some interval.

DonLakeFlyer, 2018-02-15:
QGC never expects to be in control of sending Follow Me position updates. It has no idea about any external thing which might be doing that. To me the fix for this is "don't do that"! Why would you need/want to use an external app to provide follow me updates if you are using QGC?

hamishwillee, 2018-02-15:
@DonLakeFlyer 

I was imagining perhaps a case where someone else is monitoring the flight with QGC but the actual follow me target is on a smartwatch or phone carried by the user. This would be a problem whether you had dronecore or QGC on the target user.  I'd say this is a reasonable use case, though certainly not a "core" one.

So is right solution for now to simply say "not supported"?

DonLakeFlyer, 2018-02-16:
> So is right solution for now to simply say "not supported"?

Ok, got it. Correct. Issue should stay open so it can get supported at some point.

Stifael, 2018-04-13:
> I'd say this is a reasonable use case

I agree. During debugging I use QGC to monitor the flight path. Right now i don't have any other app that can do that...

dogmaphobic, 2018-04-30:
I assume this can now be closed? (see https://github.com/mavlink/qgroundcontrol/pull/6380)

hamishwillee, 2018-05-02:
Thanks @dogmaphobic . Excellent solution.

closed 2018-05-02 by hamishwillee