mavlink/qgroundcontrol #6145: Broken UI running as non-administrator on Win10

optix2000, 2018-02-17:
- Ground station operating system.
Windows 10 Pro 1704 build 16299.248
Surface Pro 4
- QGroundControl version and build (daily, stable, self-built from source, etc.)
3.2.7 and 3.3.0. Both stable builds
- Autopilot board: Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.
N/A
- Autopilot firmware (e.g. PX4, ArduPilot, custom) and version.
N/A
- Exact steps to reproduce the problem. Starting from booting QGroundControl to when the problem occurs.
The latest update of Windows seems to have caused some incompatibilities with QGroundControl.
If I don't run QGroundControl as administrator, the UI doesn't render properly. The icons are all missing and the buttons are over-sized. All the functionality still works (assuming you can find the buttons). Choosing "GPU Compatibility Mode"  or GPU Safe Mode" doesn't change anything. Running QGroundControl as administrator makes the UI render properly again.

Running as non-admin
[image]
Running as admin
[image]

I was previously on 3.2.7, and updated to 3.3.0 to see if it would resolve the issue, which it didn't.
I'm also running with 200% text scaling in Windows.

DonLakeFlyer, 2018-02-17:
> I'm also running with 200% text scaling in Windows.

Can you change back to normal text scaling to see if that causes the problem?

optix2000, 2018-02-17:
Unfortunately no go. Though the UI does look properly sized now.
[image]

Also 200% is the default on a Surface Pro 4, given the pixel density. So even if this was the issue it's not a great workaround as it makes Windows unusable.

NaterGator, 2018-02-17:
This looks to me like the Qt SVG icon plugin is not loading normally when running without administrative privileges. Can you run as a normal user and capture the debug console output?

optix2000, 2018-02-18:
> Can you run as a normal user and capture the debug console output?

This one?
```
[D] at ..\src\QGCApplication.cc:263 - "Settings location "C:/Users/optix/AppData/Roaming/QGroundControl.org/QGroundControl.ini" Is writable?: true"
[D] at ..\src\QGCLoggingCategory.cc:99 - "Filter rules "qt.network.ssl.warning=false\n*Log.debug=false\n""
[E] at :0 - "serialnmea: No serial ports found"
[D] at ..\src\PositionManager\PositionManager.cpp:39 - "QGeoPositionInfoSourceWinRT(0x125d3c18)"
[D] at ..\src\Vehicle\MAVLinkLogManager.cc:359 - "MAVLink logs directory: "C:/Users/optix/Documents/QGroundControl/Logs""
[D] at ..\src\QtLocationPlugin\QGCMapEngine.cpp:216 - "Map Cache in: "C:/Users/optix/AppData/Local/cache/QGCMapCache300" / "qgcMapCache.db""
[E] at file:///C:/Program Files (x86)/QGroundControl/QtQuick/Dialogs/DefaultFileDialog.qml:210 - "file:///C:/Program Files (x86)/QGroundControl/QtQuick/Dialogs/DefaultFileDialog.qml:210:25: QML ListView: Binding loop detected for property "model""
[E] at file:///C:/Program Files (x86)/QGroundControl/QtQuick/Dialogs/DefaultFileDialog.qml:210 - "file:///C:/Program Files (x86)/QGroundControl/QtQuick/Dialogs/DefaultFileDialog.qml:210:25: QML ListView: Binding loop detected for property "model""
[E] at file:///C:/Program Files (x86)/QGroundControl/QtQuick/Dialogs/DefaultFileDialog.qml:210 - "file:///C:/Program Files (x86)/QGroundControl/QtQuick/Dialogs/DefaultFileDialog.qml:210:25: QML ListView: Binding loop detected for property "model""
[E] at file:///C:/Program Files (x86)/QGroundControl/QtQuick/Dialogs/DefaultFileDialog.qml:210 - "file:///C:/Program Files (x86)/QGroundControl/QtQuick/Dialogs/DefaultFileDialog.qml:210:25: QML ListView: Binding loop detected for property "model""
[E] at qrc:///qml/QGroundControl/Controls/QGCFlickableVerticalIndicator.qml:6 - "qrc:///qml/QGroundControl/Controls/QGCFlickableVerticalIndicator.qml:6: TypeError: Cannot read property of null"
[E] at qrc:///qml/QGroundControl/Controls/QGCFlickableHorizontalIndicator.qml:6 - "qrc:///qml/QGroundControl/Controls/QGCFlickableHorizontalIndicator.qml:6: TypeError: Cannot read property of null"
[!] at :0 - "QWindowsNativeFileDialogBase::shellItem: SHCreateItemFromParsingName(file:///c/Users/optix/Documents/QGroundControl/Logs)) failed (An attempt was made to reference a token that does not exist.)"
[!] at :0 - "QWindowsNativeFileDialogBase::shellItem: SHCreateItemFromParsingName(file:///c/Users/optix/Documents/QGroundControl/Logs)) failed (The operation completed successfully.)"
[D] at :0 - "QWindowsNativeFileDialogBase::onSelectionChange () 0"
[D] at :0 - "QWindowsNativeFileDialogBase::onSelectionChange () 0"
```

> Does the user account in question have something set for the environment variable QT_PLUGIN_PATH?

I don't think so. The only QT related env variable is `QT_DEVICE_PIXEL_RATIO=auto`
```
C:\Users\optix>echo %QT_PLUGIN_PATH%
%QT_PLUGIN_PATH%

C:\Users\optix>
```

NaterGator, 2018-02-18:
Okay, to confirm if I am right about this can you do the following:

1. Download [Process Explorer](https://docs.microsoft.com/en-us/sysinternals/downloads/process-explorer) from Microsoft
2. Run qgc as a normal user
3. Find QGC in Process Explorer and select it
4. Press Ctrl + L, Ctrl + D to show the lower panel and view loaded DLLs
5. See if qtsvg.dll and qtsvgicon.dll are loaded, and what path they are loaded from if so.
6. Repeat but run qgc as administrator.

This is what I expect you to see when the icons are there:
[image]

I'm not sure what you're going to get when the icons are not there, but I suspect the qtsvgicon.dll plugin will not be loaded.

optix2000, 2018-02-18:
Both seem to load the same DLLs.

Non-admin:
[image]

Admin:
[image]

optix2000, 2018-02-18:
I do notice that when running as a non admin, `QT_QUICK_BACKEND` and `QT_SCALE_FACTOR` are set, which may explain the UI size difference. These are not system or user environment variables, so I'm not sure where they're being set from.
[image]

NaterGator, 2018-02-18:
Interesting, those are not set programmatically as part of the Qt API or by us, so you are probably on to something there. The first is changing the scenegraph backend.

optix2000, 2018-02-18:
More interestingly, I can get these from the command prompt, but they aren't set from the "Environment Variables" dialogue.

```
C:\Users\optix>set
----SNIP----
QT_DEVICE_PIXEL_RATIO=auto
QT_QUICK_BACKEND=softwarecontext
QT_SCALE_FACTOR=2
----SNIP----
```

[image]

NaterGator, 2018-02-18:
They're not set in the system variables either? That's incredibly strange, I can't find a qputenv for QT_QUICK_BACKEND anywhere in Qt's source code and it isn't present in any qgc code either.

DonLakeFlyer, 2018-02-18:
> This looks to me like the Qt SVG icon plugin is not loading normally when running without administrative privileges.

Could there be missing plugins from the installer?

NaterGator, 2018-02-18:
No, they're picked up automatically by windeployqt. @optix2000 is on the right track with these environment variables he's found which are set. If I specify QT_QUICK_BACKEND=software I get the same results. He's gotta figure out where / why that is set on his system but it is not something we set and it isn't a QSG renderer plugin we ship so I'm not surprised it is failing.

optix2000, 2018-02-18:
Pretty sure this isn't a QGroundControl issue then.

A reboot fixes the issue, so I'm guessing there's another QT program that's setting it globally, which is causing issues. I'll see if I can get it to repro again for posterity. Thanks for all the help!

DonLakeFlyer, 2018-02-18:
Some other Qt app on the system? But then that doesn't make sense with respect to admin versus non-admin.

NaterGator, 2018-02-18:
It does, the QT_QUICK_BACKEND env var is in user scope, not system. Admin is a different user without that var set, so everything works in admin mode. I'm gonna go ahead and close this one.

closed 2018-02-18 by NaterGator