mavlink/qgroundcontrol #6150: Commit all UI field on mission upload

LorenzMeier, 2018-02-19:
@DonLakeFlyer You can get into a quite unexpected state when uploading a mission:

  - Add all waypoints
  - Edit one, enter a new altitude but leave the cursor in the field and do NOT press enter
  - Hit upload mission

The mission will upload with the old altitude. Can we make sure to commit all fields on hitting the mission upload button?

We found out the hard way by crashing a drone - it came in too low.

DonLakeFlyer, 2018-02-19:
You are using Stable 3.3? I fixed a ton of these already.

DonLakeFlyer, 2018-02-19:
Crap. This one looks to still be there.

DonLakeFlyer, 2018-03-12:
The fix for this still doesn't lead to a totally satisfactory user experience. Here is what happens:
* Change of mission altitude text field is un uncommitted state
* User clicks Upload button
* Upload is done before the altitude change is completed
* Altitude change triggers Prompt to ask use if they want to update mission with new altitudes
* Click yes to adjust altitudes
*  You are now in a state that your mission is dirty again. Upload button changes to "Upload Required". 

Which means you need to upload again to get new altitudes to go to vehicle. Given no-save model, there isn't an easy way to fix this ordering of how things happen. I'm going to create a new Issue with this specific use case to think about if there is some way to make this better.

closed 2018-03-12 by DonLakeFlyer