mavlink/qgroundcontrol #6151: Request: solution for resetting parameter values to startscript values

Stifael, 2018-02-21:
This request requires autopilot firmware and qgroundcontrol changes:

Issue:
Currently the default parameters are saved in xml files and are embedded as resources into qgc. Let's call these parameter `Default`. These values, however, can be overwritten by the values set in the start script. Let's call those values in the start script `Vehicle-Default` values.
During operation, the user might decide to change one of the parameter values to something else. For whatever reason he might later decide to reset that parameter to its `Vehicle-Default` value. With the current implementation, however, this is not possible and the parameter can only be reset to its `Default` value. This is unfortunate because `Vehicle-Default` and `Default` values might differ significantly.

MaEtUgR, 2018-02-21:
The same principle applies to the maximal and minimal value. It's vehicle specific e.g. what maximal speed it is able to achieve hence it would make sense to set `Vehicle-Limits`.

DonLakeFlyer, 2019-01-19:
Correct default handling is in discussion with new parameter protocol

DonLakeFlyer, 2020-11-20:
This is being dealt with in the new COMPONENT_INFORMATION spec. Although the actual solution is still not figured out.

closed 2020-11-20 by DonLakeFlyer