mavlink/qgroundcontrol #6153: Feature Request: Disable mission download on connect

mwingerson, 2018-02-21:
I want to be able to disable the mission download when a drone connects in QGC. I find it really annoying to wait if there is a large mission loaded on the autopilot. If somebody could point me in the right direction and I can help implement this change.

Thanks!
Marshall

dagar, 2018-02-25:
Take a look at the differences for a high latency connection. https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/Vehicle.cc#L236

High latency connections skip the initial mission and parameter sync.

DonLakeFlyer, 2018-03-13:
Just FYI: If you skip parameter sync you lose a bunch of QGC ui.

Skipping mission download ends up being a bit of a safety problem since you could accidentally fly a vehicle with a mission on it which you don't know is there. Would need some way to determine there is a mission on the vehicle to at least alert the user (not currently possible). Or an option to clear a mission from a vehicle any time you connect instead of downloading.

closed 2026-10-04 by github-actions[bot]