mavlink/qgroundcontrol #6161: Needing Translations for this platform?

caratzky, 2018-02-25:
Good day Sir!

I am an active Utopian Contributor for Translations to my native Language, Filipino/Tagalog. I would like to know if you need some translations for your .md and other translatable files. I've been translating files here in Github for a long time already, and I post these translations at Utopian.io for rewards. Once I manage to post at least once about your project at Utopian, more translators will come flooding here to do the same for other languages.

If you permit so, it would be a great help to me, and I'll make sure my translations are of the best quality.
Hoping to hear from you soon. God bless and more power!

Best Regards,
caratzky | Rizalino Caratao-built from source, etc.)

DonLakeFlyer, 2018-03-13:
@dogmaphobic Is this a language we have posted?

dogmaphobic, 2018-03-13:
I'm open to any language where there is demand. However, we do not support Utopian and have no plans to do so.

diegopucci, 2018-04-10:
Hello @dogmaphobic . CEO of Utopian.io here. It is sad to read your comment above. We try hard to consider each and every feedback. Especially project owners are important for our community to grow. Please consider sharing with me feedbacks about Utopian and what you think it is wrong and may be improved. Thanks for your time

closed 2018-04-30 by DonLakeFlyer