mavlink/qgroundcontrol #6163: About  Google satellite map in China；

MyOnlyOnlySunshine, 2018-02-27:
[image]

Click Google satellite map failed to load the following print log information, but other types of Google maps normal, may I ask what is the reason, thank you!

dogmaphobic, 2018-02-27:
Do they work from a regular browser? That is, if you where to launch a browser and go to:

https://www.google.com/maps/@31.1279242,121.4893596,525m/data=!3m1!1e3

Do you see the map?

MyOnlyOnlySunshine, 2018-02-28:
yes ， I can see the map from the link that you support .

dogmaphobic, 2018-06-20:
I have no way to work on this. If someone in China can take a look, great. Otherwise, I'm closing it.

closed 2018-06-20 by dogmaphobic