mavlink/qgroundcontrol #6167: Bad user experience on windows touch screen tablet

feemi, 2018-02-27:
Panasonic tough pad touch screen is working pretty good normally. But in app tappings pinchings and slidings are not working smoothly. 

Survey planning is almost impossible. 
Sliding to confirm is really hard to achieve.
Dropdowns needs long tap or double tap to open.

----
General Information
- Panasonic ToughPad FZ-G1 Windows 10 Pro
- v3.3.0
- Pixhawk 2
- PX4

DonLakeFlyer, 2018-03-13:
@NaterGator Do you have any touch screen windows devices? I don't have anything to test whether this is a general problem or not.

NaterGator, 2018-03-13:
I have a Surface Pro 3 I can try it out on. If memory serves I have been a bit unimpressed with Qt's multitouch gesture handling on Windows before.

feemi, 2018-03-13:
I realised that experience got worse when i update windows 10. It was almost impossible to use it.  When i reverted the windows updates experience is acceptable with pen. But still moving survey plan polygons with finger is frustrating.

feemi, 2018-10-23:
Hi Don, The experience got better after qt update. We have still some issues with dropdown tapping. but pinching map is way better now.

DonLakeFlyer, 2018-10-24:
> We have still some issues with dropdown tapping. but pinching map is way better now.

Can you give a specific example?

DonLakeFlyer, 2019-01-19:
I'll make a swing prior to 3.5 stable with windows touchscreen. If you have a specific dropdown case I can look at can you create a new issue.

closed 2019-01-19 by DonLakeFlyer