mavlink/qgroundcontrol #6169: Add "radar" view using DISTANCE_SENSOR messages from autopilot

rmackay9, 2018-03-01:
ArduPilot supports object avoidance using various rangefinders (ZED 3D camera, 360 lidar, single lidar).  As part of this feature, the AP_Proximity library sends DISTANCE_SENSOR messages with the appropriate distance and orientation to the ground station.

This request is to add a "radar" like view so the operator can see the distances to nearby objects.

DonLakeFlyer, 2018-03-12:
@dogmaphobic Did you already add something like this for the Typhoon?

dogmaphobic, 2018-03-12:
Yes, I did something quite simple using the `MAVLINK_MSG_ID_DISTANCE_SENSOR` message.

DonLakeFlyer, 2018-03-12:
Can that be brought upstream and support all the horizontal values?

dogmaphobic, 2018-03-12:
Not exactly and as it is, it's not that useful as that was made for a particular type of sensor working in a predictable way. QGC would need something more abstract. There are messages giving sensor values but nothing to enable/disable it for instance. If you have something but it's disabled, you will get the message but nothing ever changes, which is not too useful. You need to know it's active (and a way to make it active/inactive). Martina has also been working on some multidimensional sensors (she has done a boatload of work on that) and there are new messages for those. We need a plan so I know what to implement and how.

For example:
https://github.com/PX4/Firmware/pull/8896

DonLakeFlyer, 2018-03-12:
> If you have something but it's disabled, you will get the message but nothing ever changes, which is not too useful.

That would just seem to be a broken/studid firmware implemention? No?

DonLakeFlyer, 2018-03-13:
@rmackay9 Is there a way to simulate this with ArduPilot SITL for testing/implemenation?

DonLakeFlyer, 2018-03-21:
Hints on testing with SITL:
I need to check but normaly if you activate the sonar on sitl, you should have it on distance sensor msg. With mavproxy : load Tools/autotest/default_params/copter_rangefinder.parm or something like that. Then relaunch Sitl and you should have sonar and the distance sensor message with orientation pitch 270

DonLakeFlyer, 2018-04-05:
Hmm, I missed the all level and up/down sensors. I'll add to the pull.

DonLakeFlyer, 2018-06-20:
Ok, so the values are all plumbed through and can be displayed. I'll add a better display in the next release.

DonLakeFlyer, 2020-11-20:
There is a radar view in the Fly View now

closed 2020-11-20 by DonLakeFlyer

ZePedro80, 2021-07-15:
Hello! I try to ask for the DISTANCE_SENSOR message (current_distance parameter) but it shows big random values most of the time (very occasionaly the values make sense). In QGroundControl the values are ok, so I don't know why I am receiving wrong values in my own program. Every other parameter from any message ID are perfectly received, but this one isn't.
Any of you have an idea of what it could be?