mavlink/qgroundcontrol #6170: Elevation/Terrain data not working anymore

philipoe, 2018-03-01:
The elevation/terrain data retrieval process does not work anymore.

OS: Ubuntu 16.04 (actually tested on mine and a colleague's computer on which it worked before)
QGC: Master, #6711810 (February 24th 2018)
Autopilot: Pixhawk (but tested in SITL or when the autopilot was not even connected)
How to reproduce: Set a number of waypoints. Not terrain data is shown anymore.

Debugging: I don't know the AIRMAP api and the respective code in QGC well, but tried to debug and at least got the following: The request sent to the airmap servers looks as follows:

[image]

The response from the airmap servers is then a html document that has the content "Error, cannot GET SRTM1 ele". So I assume something in the airmap API changed (although i could not find good documentation on how to retrieve elevation data from their APIs).

@DonLakeFlyer @birchera You definitely have a much better insight into the airmap API.

birchera, 2018-03-13:
The elevation API seems broken. I contacted Airmap and will update here once I get a response.

DonLakeFlyer, 2018-03-13:
Thanks folks for looking into this.

birchera, 2018-03-15:
No response yet.

DonLakeFlyer, 2018-03-15:
I"ve pinged them on a direct slack channel that I have to them. They've ackknowledged but no solution yet.

birchera, 2018-03-16:
Thanks for fixing. Will push the necessary changes for offline elevation data shortly.

DonLakeFlyer, 2018-03-16:
FYI: I'm having to rework the correct Terrain.h/cc code. So you should wait until I push that through. Current code was a mess with respect to adding additional query types. Like heights along a path for example.

closed 2018-03-22 by DonLakeFlyer