mavlink/qgroundcontrol #6171: QGC 3.3: GeoTagging does not work.

tubeme, 2018-03-01:
Hello guys,

We are trying to geotag images for import in Pix4d but without success.

We select the log, we select the image folder and we use the default TAGGED folder. Then the progress bar shows that the images are geotagged. QGC created the TAGGED folder but it is empty. We checked if there are any geotags on the images and there is none shown in file properties in windows, neither there are geotags seen in adobe properties.

Strangely if we repeat the operation QGC says the images are allready geotaged, so somehow it sees some kind of hidden geotags in the files.

For a Pix4D we need a text file with coordinates  and image numbering, that is supposed to be automatically generated by the tagging software. There is no such TXT file either.

Probably something is wrong and we really cannot understand what. It is pretty simple and straight forward process but it does not work.

We tried the 3.2 and 3.3 version with the same results. We fly with PX4 1.6.5.

Any help will be appreciated.

tubeme, 2018-03-01:
Forgot to mention we use UTC in the camera and in the LOG files in order to have both in sync.

dagar, 2018-03-05:
I don't suppose you could share a dataset for debugging?

tubeme, 2018-03-05:
What you mean by data set? I can share the log with the images taken from the flight. Is it of any help?

dagar, 2018-03-06:
Yes the images + corresponding log that you were trying to tag.

tubeme, 2018-03-06:
Here you go.

https://ndox.abv.bg/download?id=8dec10e483

We also tried to change the Windows time to UTC and then run the QGC. The same result.

tubeme, 2018-03-09:
@dagar @DonLakeFlyer Everithing OK with the download? Any ideas?

dogmaphobic, 2018-03-15:
I'm looking into this...

dogmaphobic, 2018-03-16:
First, I have to start saying I know nothing about ulog other than what it is. Looking through the code, it seems there is no camera capture data in that log file. At least the code doesn't find any. I have no easy way to parse the file myself to check. I tried both logs.px4.io and MAVGAnalysis but it seems neither one report camera triggers.

Before spending a great deal of time learning how to parse ulog files (so I can validate the existing code within QGC) are you certain this log file does in fact contain these camera triggers? 

With that said, the geotag code in QGC is quite inefficient and not too user friendly but that's something for a separate issue.

SinigerVTL, 2018-03-16:
Thanks Gus,

I see now. That was our presumption as well that there are no triggers in the log file. I don't know how we sense it but that was our first thought... We will make another flight and collect more data. So we stop here for now.

tubeme, 2018-03-16:
Thanks @dogmaphobic,

I see now. That was our presumption as well that there are no triggers in the log file. I don't know how we sense it but that was our first thought... We will make another flight and collect more data. So we stop here for now.

DonLakeFlyer, 2018-04-30:
This should be all fixed up now as related to discussion and fixes linked to #6234

closed 2018-04-30 by DonLakeFlyer