mavlink/qgroundcontrol #6183: udp link problem

yl5006, 2018-03-07:
qgroundcontrol link for simulater in gazebo using different computer host 。
，the qgroundcontrol can not get full link for simulater in gazebo， pcanmeters can not  pull down,and  the vehicle do not respond qgrodncontrol command.

yl5006, 2018-03-07:
can not get remote QHostAddress

DonLakeFlyer, 2018-03-13:
Going to need some explanation of the problem which is less cryptic than above. As far as I can tell Gazebo connections are working fine. Detail out the steps to reproduce.

closed 2018-03-14 by yl5006

yl5006, 2018-03-15:
I use QT 5.9.3. The Debug consol show  Error writing to QHostAddress("") 2682

yl5006, 2018-03-17:
@DonLakeFlyer I change to QT5.9.1 this problem do not appear

DonLakeFlyer, 2018-03-17:
I don't know much about networking problems. Maybe @dogmaphobic has ideas.

DonLakeFlyer, 2018-03-17:
First step would be use a pre-built version, not a version you built yourself. If you build it yourself we have no idea would could go wrong.

yl5006, 2018-03-20:
yes。 I cost a lot of time to find the wrong。I change ide to QT5.9.1  ，and changed UAV simulate Firmware start script which I forget to change it（I used to modify it before ） ，then it go well。
here my mavlink start script 。why groundstation need so fast msgs？so i comment out others
mavlink start -x -u 14556 -r 4000000 -t 192.168.101.72
#mavlink start -x -u 14557 -r 4000000 -m onboard -o 14540
#mavlink stream -r 50 -s POSITION_TARGET_LOCAL_NED -u 14556
#mavlink stream -r 50 -s LOCAL_POSITION_NED -u 14556
#mavlink stream -r 50 -s GLOBAL_POSITION_INT -u 14556
#mavlink stream -r 50 -s ATTITUDE -u 14556
#mavlink stream -r 50 -s ATTITUDE_QUATERNION -u 14556
#mavlink stream -r 50 -s ATTITUDE_TARGET -u 14556
#mavlink stream -r 50 -s SERVO_OUTPUT_RAW_0 -u 14556
#mavlink stream -r 20 -s RC_CHANNELS -u 14556
#mavlink stream -r 250 -s HIGHRES_IMU -u 14556
#mavlink stream -r 10 -s OPTICAL_FLOW_RAD -u 14556