mavlink/qgroundcontrol #6185: Params not loading Android app via XBee

Patnetpatnet, 2018-03-07:
Using Bluetooth or USB, the parameters don't fully load into the app. It kind of times out. As the blue line progress it slows at about 80% then slowly completes. All baud rates are 57600. Fails with Mavlink 1 and Mavlink 2. FC is Px4 hardware. App is 3.3.0. Same issue with Daily Build version.

DonLakeFlyer, 2018-03-13:
Turn on ParameterManagerVerbose1Log (https://docs.qgroundcontrol.com/en/SettingsView/console_logging.html) and attach the log output here.

Patnetpatnet, 2018-03-13:
Thanks Don. Attached. UI error at end is “…was unable to retrieve the full set of pararemters…” in a yellow dialog box.

[Paramloadlogqgc.txt](https://github.com/mavlink/qgroundcontrol/files/1808273/Paramloadlogqgc.txt)

DonLakeFlyer, 2018-03-13:
The important log entry is here:
[E] at /home/travis/build/mavlink/qgroundcontrol/src/FactSystem/ParameterManager.cc:1105 - ""V:1" The following parameter indices could not be loaded after the maximum number of retries:  "405, 448""

This means that QGC asked the vehicle to send the parameter for index 405 and 448 five seperate times with no response. After 5 retries it gave up. Not much QGC can do at that point, sorry. QGC is doing all the right things.

Possibility is extremely noisy link which is dropping lots of packets. Some sort of XBee setup problem? Don't know much about XBee's so can't help much there.

closed 2018-03-13 by DonLakeFlyer