mavlink/qgroundcontrol #6189: Auto-Connects to all saved Comm links

ryanjAA, 2018-03-11:
Using QGC 3.3.0 release on a mac, once started QGC tries to connect to every comm device that is saved (and that has auto-connect enabled), even if they are not plugged in. Doesn't do this with previous versions. I have 7 different comm links saved. Does it with serial and tcp links.

ryanjAA, 2018-03-12:
@dagar

DonLakeFlyer, 2018-03-12:
What exactly is the bad behavior? Errors popping up you didn't get before?

ryanjAA, 2018-03-13:
Automatically connects to every saved comm-link (ones that are not plugged in)

[image]

DonLakeFlyer, 2018-03-23:
@ryanjAA I'm trying to remember how it worked before. If auto-connect was checked what did that mean in previous builds?

ryanjAA, 2018-03-23:
Autoconnect is present in the previous builds in the Preferences -> General, then there is a box for choosing which type you want it to auto connect to. Previous to this version even if they were all selected to connect to and you had 10 serial links saved, it would only connect to ones that were plugged in. I just loaded up QGC 292 and all my links are still saved but it doesn't try to connect to them unless I plug one in. Conversely, I just loaded up todays daily build and it still tried to connect to all of them.

DonLakeFlyer, 2018-03-25:
Ah, ok. I was looking at the wrong autoconnect setting.

DonLakeFlyer, 2018-03-28:
Hmm, I can't repro on Daily or Stable.

DonLakeFlyer, 2018-06-20:
Is this still happeing in latest daily builds

DonLakeFlyer, 2018-06-20:
Reopen if still problems

closed 2018-06-20 by DonLakeFlyer