mavlink/qgroundcontrol #6191: Set RC to Param not working in QGC v3.3.0

larryyonge, 2018-03-12:
Selecting "Advanced settings" -> "Set RC to Param..." in Parameter Editor is not working in QGC 3.3.0 with PX4 v1.7.3 on Pixhawk, tested on Ubuntu 16.04.  In the Set RC to Param Dialog window, "Center value" and "OK" are grayed out and "Waiting for parameter refresh..." is displayed.  "Set RC to Param" works correctly in QGC v3.2.7.

[image]

closed 2018-03-21 by DonLakeFlyer

BazookaJoe1900, 2018-06-06:
Has this been fixed on 3.3.2?
not for me, in windows

DonLakeFlyer, 2018-06-06:
It is fixed in daily builds. It will not be fixed in 3.3.2

Molecuman, 2019-09-04:
Is not working for me on the 3.5 version

DonLakeFlyer, 2019-09-04:
> Is not working for me on the 3.5 version

@Molecuman Please open a new issue with some useful details as to how exactly it is "not working"

Molecuman, 2019-09-04:
The same thing of this thread

DonLakeFlyer, 2019-09-04:
> In the Set RC to Param Dialog window, "Center value" and "OK" are grayed out and "Waiting for parameter refresh..." is displayed.

This is what you are seeing?

Molecuman, 2019-09-04:
No. The RC to parameter function is not showing up

DonLakeFlyer, 2019-09-04:
I just tried 3.5.4 and it's working fine for me:
[image]

Molecuman, 2019-09-04:
Maybe is my system or the QGC Windows version. The button is there, is not
gray. The screenshot just shows the button highlighted but after pushing
it, nothing happens

[image: image.png]

WhiteBearSpirit, 2020-04-13:
Same problem: pressing 'Set RC to Param' is now showing any dialog box. QGroundControl 4.0.5, Windows 10.

DonLakeFlyer, 2020-04-14:
> pressing 'Set RC to Param' is now showing any dialog box. QGroundControl 4.0.5, Windows 10.

This feature is missing from the 4.0 releases. I'll add it back soon.

DonLakeFlyer, 2020-04-14:
Also until then Set RC to Param just sets up a number of parameters with a nice ui. It all can be done manually through the parameter editor.

DonLakeFlyer, 2020-04-14:
> Also until then Set RC to Param just sets up a number of parameters with a nice ui. It all can be done manually through the parameter editor.

Sorry. That is incorrect. This is a command not parameters. So no way to do it without GCS support.

DonLakeFlyer, 2020-04-15:
@Molecuman I added it back: https://github.com/mavlink/qgroundcontrol/pull/8657. Can you help with testing it out