mavlink/qgroundcontrol #6199: Hi All, Is it possible to customize or change theme of map used in QGC, precisely want to change theme of Google map used for QGC. Google is providing javascript based api but don't know how to use that with C++ based code as used in QGC.

arun2204, 2018-03-13:
This system is for posting bugs or feature requests ONLY. For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

----
When posting bug reports, include the following informaiton:
- Ground station operating system.
- QGroundControl version and build (daily, stable, self-built from source, etc.)
- Autopilot board: Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.
- Autopilot firmware (e.g. PX4, ArduPilot, custom) and version.
- Exact steps to reproduce the problem. Starting from booting QGroundControl to when the problem occurs.

**All this template text should be deleted before creating your issue.**

arun2204, 2018-03-13:
Hi All, Is it possible to customize or change theme of map used in QGC, precisely want to change theme of Google map used for QGC. Google is providing javascript based api but don't know how to use that with C++ based code as used in QGC.

DonLakeFlyer, 2018-03-13:
QGC uses URL based map tile query. Changing to javascript would be a huge undertaking.

closed 2018-03-13 by DonLakeFlyer