mavlink/qgroundcontrol #6202: QGround Control RTSP Video is killing me.

Rickyg32, 2018-03-13:
Well this is killing me. I can’t get this function to work properly. in QGroundControl
I can have one test camera with a string like
rtsp://admin:123456@192.168.0.98:554/mpeg4 This works

The second test source should use a similar string
rtsp://root@admin@192.168.0.90/mpeg4/media.amp This does not

The log fills with these errors
(x86)/QGroundControl/QtQuick/Controls/Private/BasicButton.qml:189:5: QML MouseArea: Cannot anchor to an item that isn’t a parent or sibling."
[E] at file:///C:/Program Files (x86)/QGroundControl/QtQuick/Dialogs/DefaultFileDialog.qml:210 - “file:///C:/Program Files (x86)/QGroundControl/QtQuick/Dialogs/DefaultFileDialog.qml:210:25: QML ListView: Binding loop detected for property “model””
[!] at :0 - “QWindowsNativeFileDialogBase::shellItem: SHCreateItemFromParsingName(file:///c/Users/RichardGardner/Documents/QGroundControl/Logs)) failed (An attempt was made to reference a token that does not exist.)”
[!] at :0 - “QWindowsNativeFileDialogBase::shellItem: SHCreateItemFromParsingName(file:///c/Users/RichardGardner/Documents/QGroundControl/Logs)) failed (The operation completed successfully.)”
[D] at :0 - “QWindowsNativeFileDialogBase::onSelectionChange () 0”

Rickyg32, 2018-03-13:
This is driving me nuts I can stream on an device either of the RTSP streams but not from QGroundcontrol. Only one works.
I cannot no matter what I do get this RTSP stream to show up on the screen.
Now the log fills with this error
[!] at ..\src\VideoStreaming\VideoReceiver.cc:544 - "Resource not found."
[D] at :0 - "QWindowsNativeFileDialogBase::onSelectionChange () 0"
[D] at :0 - "QWindowsNativeFileDialogBase::onSelectionChange () 0"
[!] at ..\src\VideoStreaming\VideoReceiver.cc:544 - "Resource not found."
[!] at ..\src\VideoStreaming\VideoReceiver.cc:544 - "Resource not found."
[!] at ..\src\VideoStreaming\VideoReceiver.cc:544 - "Resource not found."
[!] at ..\src\VideoStreaming\VideoReceiver.cc:544 - "Resource not found."
[!] at ..\src\VideoStreaming\VideoReceiver.cc:544 - "Resource not found."
[!] at ..\src\VideoStreaming\VideoReceiver.cc:544 - "Resource not found."
[!] at ..\src\VideoStreaming\VideoReceiver.cc:544 - "Resource not found."

DonLakeFlyer, 2018-03-13:
@dogmaphobic @jaxxzer Any ideas?

jaxxzer, 2018-03-13:
@Rickyg32 what is the format of these streams? The url indicates it may be MPEG4, QGC will only take H.264 streams.

Rickyg32, 2018-03-13:
Then thats the problem. I am pretty sure its only mpeg4 streams available to me. damn

DonLakeFlyer, 2018-03-13:
@jaxxzer Is there some way to add some qDebug output ot something to indicate the format we are getting is incorrect?

jaxxzer, 2018-03-13:
Yes, and there may be something indicative of this already. The logs @Rickyg32 provided are truncated.

DonLakeFlyer, 2018-03-13:
qgcApp()->showMessage("...") when it detects a bad stream format?

jaxxzer, 2018-03-13:
Sure, not positive right now if we will be able to identify failure reason on that level, but "probably".

Rickyg32, 2018-03-13:
What logging do you need. I would be glad to grab a log and post it some where if required. However if the tech doesn't support Mpeg2 or HTTP video then thats why its not working for me. I didn't think..doh that RTSP streams would differ due to compression

jaxxzer, 2018-03-13:
If you could provide the entire log that is produced after starting up qgc with the stream name entered the previous run, that would be good. And let us know the exact stream url you use for this particular log.

Make sure videoreceiver log is enabled as well.

Rickyg32, 2018-03-13:
[App Log.txt](https://github.com/mavlink/qgroundcontrol/files/1808479/App.Log.txt)
Here is the log. Let me know if you want more.
The RTSP url is this
rtsp://192.168.0.90/mpeg4/media.amp

Rickyg32, 2018-03-13:
[App Log 2.txt](https://github.com/mavlink/qgroundcontrol/files/1808486/App.Log.2.txt)
Another log with a different RTSP url
rtsp://root:admin@192.168.0.90/mpeg4/media.amp
Whats odd is I am sure the other camera isn' t H.264 Will post that log next

Rickyg32, 2018-03-13:
[App Log 3.txt](https://github.com/mavlink/qgroundcontrol/files/1808492/App.Log.3.txt)
Hre is a log of a working camera. the app was not restarted so there is stuff at the start of the log from teh previous camera.
RTSP URL is rtsp://admin:123456@192.168.0.98:554/mpeg4

jaxxzer, 2018-03-13:
This situation can be handled better.

Rickyg32, 2018-03-13:
The situation...jaxxzer you need to elaborate..do you mean the on going dialog, the error condition or the general state of the world

jaxxzer, 2018-03-13:
The code can handle unexpected formats more gracefully.

Rickyg32, 2018-03-13:
I see. Well it doesn't seem to be, it gave me no hit. What sucks is Mpeg4 video streamers are cheap and easy to find. H.264..not so much. so having the ability to support Mpeg4 would be handy

Rickyg32, 2018-03-13:
So any plans to perhaps add the ability or at a minimum update the code to address the error a bit nicer

Rickyg32, 2018-03-15:
I guess there are no plans then. lol

DonLakeFlyer, 2018-03-15:
I think @jaxxzer Is going to look into it. @jaxxzer Can you enter a separate issue for that and close this one?

Rickyg32, 2018-03-16:
Thanks Guys. I love this app and want to take full advantage of the video components as I am sure most do. With out having to spend hundreds on special hardware.

Awesome

DonLakeFlyer, 2018-03-16:
FYI: I was just talking about detecting incorrect stream formats. Don't know what it takes to support additional streams.

dogmaphobic, 2018-03-16:
> Mpeg4 devices are cheap and cheerful

I have yet to see a camera using it. Everything I have, from dirt cheap to expensive stream using h.264. What camera are you talking about?

Rickyg32, 2018-03-16:
Camera..no. I have a ground station that has a video in for my FPV RX, it routes the video to an amp and splitter. This in-turn routes the video to a small video webserver. This webserver serves up the camera video via RTSP, I want to use QGroundcontrols Video function to display the video stream. I have never seen an FPV camera that sends video in H.264. 
https://www.youtube.com/watch?v=KIekc-Ux-LU&t=8s
This video shows you what I am doing, but I am wanting to improve on the video function. This requires a special ActiveX control in a browser. The QGroundContol eliminates that need if your using RTSP, but currently needs H.264 compression. If it supported Mpeg4.
Hope that helps make sense of it.

Rickyg32, 2018-03-26:
today I discovered my Video router in my groundstation is H.264 compliant it just wasn't configured to do so. In the words of Bart Simpson......DOH

Works great now.

DonLakeFlyer, 2018-03-27:
No plans right now to support Mpeg4

closed 2018-03-27 by DonLakeFlyer

maZZiveAttack, 2020-05-27:
Ok. Just an idea. If you want people to use Maverick, and people do, then the instructions need to be readable and understandable so someone without a clue can follow them. But most of all, they need to work. 
I'm using a Raspberry Pi 3 Model B and I would like to have a Camera stream to either QGC or MP. I'm following the instructions as best as I can but it's not working. I'm sure you are all busy people but if you're going to give instructions then why can't you lay them out with examples, screenshots, bullet points, Step 1. ... Step 2. Why is it so difficult?

DonLakeFlyer, 2020-05-27:
If you are having problems getting video to work there is a Slack video streaming channel which has video folks on it that may be able to help.