mavlink/qgroundcontrol #6204: Mission Terrain Support without firmware terrain following

DonLakeFlyer, 2018-03-13:
This is to start the discussion on supporting altitude above terrain in missions. This specific discussion assumes that the firmware does **not** have support for MAV_FRAME_GLOBAL_TERRAIN which would require some sort of terrain tile push to the vehicle. The assumption is that this is solely done on the ground station by setting appropriate mission item heights and/or possibly adding new wayoints to handle a terrain problem in between two added waypoints.

DonLakeFlyer, 2018-03-13:
@birchera @philipoe @AndKe @antiheavy @sanderux FYI. Does this statement above make sense as a problem statement?

philipoe, 2018-03-13:
Yes. As a bare minimum, the user should be warned to indicate if at any position along a simplified path between the waypoints (e.g. straight line path between two position WPs) the vehicle is close to colliding with terrain. 

Visualizing the terrain in a nicer way such that the user can see exactly where a terrain collision might occur is of course harder than just outputting a warning. Possible thoughts: 1) visualizing the terrain along the path in "google earth style" in the terrain altitude widget (although this is probably too much information in such a small window) or 2) changing the path's color in the map view as a function of distance to terrain (e.g. just marking it in red if a collision could happen)?

AndKe, 2018-03-13:
One scary scenario is that aircraft may fail to fetch elevation data during mission.  , be it due to GCS link, or internet access.

Planning a mission that does not use MAV_FRAME_GLOBAL_TERRAIN may be useful to closely verify/see the result, and making sure the aircraft will have all necessary data.

On the other hand, if we can be sure the aircraft have the data needed, and/or verify *any* mission's elevation would be an even greater tool (as moving an mission item would instantly account for the elevation if the aircraft were using such navigation mode)
 
AP2 has the "mission elevation" widget, in complex terrain, it's very useful.

and should be visualized right in the:
[image]

Antiheavy, 2018-03-13:
I posted a "feature request" issue about this a while back.  I'll add this to the discussion:  https://github.com/mavlink/qgroundcontrol/issues/5524  

> 2) changing the path's color in the map view as a function of distance to terrain (e.g. just marking it in red if a collision could happen)?

I've used systems in the past that worked like this and found it a good method.  Simple.

I've also found toggle-able map layers useful in the past for this sort of thing. For example color gradient topological maps with altitudes relative to Home.

How does interaction with the firmware to prevent takeoff (or prevent auto mode) work here.  Seems like that could be a hole.

Don't go down the 3D graphics rabbit hole....

DonLakeFlyer, 2018-03-13:
> How does interaction with the firmware to prevent takeoff (or prevent auto mode) work here. Seems like that could be a hole.

When using this feature all mission items which end up being absolute altitudes. Hence I would think everytying should work fine.

DonLakeFlyer, 2018-03-13:
> AP2 has the "mission elevation" widget, in complex terrain, it's very useful.

When AP2 draws that line do you know if it is really capturing terrain data which is in between the waypoints? From what you drew it seems like it is. Just want to verify. Or is it just the same as QGC bar graph thing, but instead it uses a line graph?

DonLakeFlyer, 2018-03-13:
@philipoe I have someone funding me for terrain support in Corridor Scan. My plan is to use that as impetus to drive me to add full terrain support everywhere else as well.

DonLakeFlyer, 2018-03-13:
@philipoe My guess is that at some point I could use help with supporting the return of elevation data on a line between two points. I'm pretty sure airmap has a direct API for that. But then is going to need to be supported in the offline caching code as well once that gets in.

DonLakeFlyer, 2018-03-13:
I have a feeling a terrain api which reports back the highest terrain altitude within a polygonal area may also be required.

philipoe, 2018-03-13:
@DonLakeFlyer Great & agreed to your last three points. Let me now when & where you need help as we also have a bit of funding for implementing something like this. What is our/your timeline on this? On our side, we'd need that within 4-5 weeks. 

@AndKe Yes, exactly such a plot would be great. Of course, potential problems are how to 1) plot missions with say >20 WPs 2) plot survey items and 3) missions where the distances between waypoints are very different and we'd thus have steep terrain gradients. But I think those issues can be resolved when we actually work on this.

AndKe, 2018-03-13:
@philipoe 
1: like I mentioned before in other threads, there's a need to "explode" surveys, preferably not by saving & loading them as .txt file.
2: >20 items is already taken care of , the mission box does scroll.
3: by using color on the gradient, even a narrow peak should be visible.  
- at best , the X axis of the graph should be easy to toggle between "compact", and "proportional"   - where in compact mode, mission items are close together, while proportional sets distance by distance from previous mission item with position.
-mission items without position, could be placed by perceived execution order.
example , "5"(waypoint) , below is "6" (cam trigger off) , .. then to the right: "7" (wp) ,and below that 8(cam trigger on)  and so on ,(reducing horizontal length, and visualizing commands executed right away.)
[image]

sanderux, 2018-03-13:
@DonLakeFlyer I am pretty sure airmap offers the terrain altitude between 2 points. For fixed wing this would indeed mean setting intermittent waypoints on the extremes. Given that a fixed wing has a known maximum climb and descend rate it would be fairly easy to calculate these waypoints. perhaps some threshold should be set to prevent too many waypoints.

sanderux, 2018-03-13:
@Antiheavy what does the APM interface look like when 2 waypoints are 20km apart?

AndKe, 2018-03-14:
@sanderux "not good" is the answer, - if you have a takeoff and few items before 80 mission items 20km away, the few are to the left, the 80 are squished to the right.
Still, even with that tool that does not scale/scroll  it's easy to test it:
I save the mission to file, delete the transport part, verify the rest/grid (that is now evenly distributed)

DonLakeFlyer, 2018-05-01:
Terrain following Surveys are starting to work. May get into daily builds in a couple days or so:
[image]

kikislater, 2018-05-01:
Will be 3d view available when finished to plan ?

DonLakeFlyer, 2018-05-01:
This is a Save to KML and then loaded into Google Earth. No 3d viewing in QGC.

kikislater, 2018-05-01:
Yes understand but I know that Qt5 could bring 3d view that's why I asked ...

AndKe, 2018-05-01:
Now THAT  (3D) would be seriously sexy :)  - and useful for mission elevation checking, not to mention the flight may be viewed more like with mavcesium...

closed 2018-06-18 by DonLakeFlyer