mavlink/qgroundcontrol #6230: menu icon not shown correctly

lijieamd, 2018-03-17:
[image]

the top menu icons are not shown

i'm using the latest QGroundControl.AppImage from here:
https://s3-us-west-2.amazonaws.com/qgroundcontrol/latest/QGroundControl.AppImage

OS: ubuntu16.04LTS

DonLakeFlyer, 2018-03-22:
Working fine for me on same Ubuntu version. My guess would be something with your OpenGL driver.

@dagar Do you run Ubuntu? Could you double check a daily build?

DonLakeFlyer, 2018-04-30:
@lijieamd Are you still seeing this? have you look at updating your OpenGL driver?

lijieamd, 2018-05-01:
@DonLakeFlyer I found that all worked fine after connected to px4 board a few days ago, so i guess if there something wrong in qgroundcontrol initialization process?

ok, i'll updating my opengl driver and try again

lijieamd, 2018-05-08:
sorry, it's just my guess.
this problem has not happen again after i connected to px4 board hardware.

DonLakeFlyer, 2018-05-08:
Ok, closing since working correctly nopw.

closed 2018-05-08 by DonLakeFlyer