mavlink/qgroundcontrol #6234: geo tagging log file parsing error

mathias-h, 2018-03-18:
- Ground station operating system.
windows 10 and mac osx
- QGroundControl version and build (daily, stable, self-built from source, etc.)
3.3.0
- Autopilot board: Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.
pixhawk 1
- Autopilot firmware (e.g. PX4, ArduPilot, custom) and version.
px4 1.7.3 (50bd148f)

when we try to geotag images it fails and gives log parsing failed - tagging cancelled error.
we have had this working earlier, with some other log files, which are taken with the same px4 and qgroundcontrol version. 
for some reason the file names of the logs are different in the new ones. on the sd card the logs are placed in a folder with the date of the log and the filename is the time. in the old ones the filename is the date and the time all in one folder.

new log: https://logs.px4.io/plot_app?log=7f68c221-79f7-4c88-9ed5-8fb415e539e3 (not working)
old log: https://logs.px4.io/plot_app?log=1af894e7-1899-457f-9fde-363160a7c09c (working)

DonLakeFlyer, 2018-03-19:
@dogmaphobic Could you take a look?

dogmaphobic, 2018-03-19:
Yup

tubeme, 2018-04-03:
I have the same problem with 
QGC 3.3 windows 
PX4 1.7.3 stable
AUAV x2.1

[image]

I armed but did not fly. It had a 3d lock.

I used the manual trigger to get the images.

Here is the log: 
[13_34_37.zip](https://github.com/mavlink/qgroundcontrol/files/1871734/13_34_37.zip)

Here are the images:
https://drive.google.com/open?id=1AKOBvaTR-fu96Wu7jPgubZWgIHEheV9q

DonLakeFlyer, 2018-04-04:
@mathias-h Problem is a bug in ulog parsing. I have a fix I think.

@tubeme This log file has camera_trigger logs instead of camera_capture logs. The log parser is looking for camera_capture logs. Not sure if the other log can be made to work or not. @birchera Any ideas?

tubeme, 2018-04-04:
@DonLakeFlyer If there is a difference in the log between capture and trigger I really don't understand why shouldn't I've been able to tad a manually triggered images.... It should be the same in my opinion and work both ways.

DonLakeFlyer, 2018-04-04:
> If there is a difference in the log between capture and trigger I really don't understand why shouldn't i been able to geotag manually triggered images

I would assume as well but I know next to nothing about how this code works or the log file format. Hopefully @birchera can answer questions.

tubeme, 2018-04-04:
Also Lucy can you get me flight for the return from Zhengzhou to Beijing?

Vasko

mhkabir, 2018-04-07:
@tubeme you need to be running the camera feedback driver on the PX4 side to generate camera_capture from camera_trigger messages.

mhkabir, 2018-04-07:
https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rcS#L448

DonLakeFlyer, 2018-04-07:
@mhkabir So camera_trigger is required then to geotag. I assume because camera_trigger has no lat/lon information in it?

mhkabir, 2018-04-07:
@DonLakeFlyer Yep.

DonLakeFlyer, 2018-04-07:
Ok, so this Issue is fixed then. I am going to make the error you get better if there are no camera_feedback log items.

DonLakeFlyer, 2018-04-07:
Also bring master fixes back to 3.3

closed 2018-04-08 by DonLakeFlyer

tubeme, 2018-04-09:
I cannot seem to find CAM_FBACK_MODE in 1.7.3? Is it in the master?

tubeme, 2018-04-09:
@mhkabir @DonLakeFlyer I tried the master just now and I cannot find CAM_FBACK_MODE parameter although I see it in the PX4 Parameter Reference page...

mhkabir, 2018-04-09:
@tubeme what board are you using? The `camera_feedback` module should be built for all FMU variants.

tubeme, 2018-04-09:
@mhkabir AUAV X2.1 Probably it is not for my board...

mhkabir, 2018-04-09:
Please add it to the cmake configuration for that board,  and everything
should work :)

tubeme, 2018-04-09:
@mhkabir  I cannot I don't have compiling environment. Can somebody makes the correct cmake for the AUAV X2.1?

MR9090, 2020-03-05:
Hi everyone i'm in trouble, i got error matching with "tagging cancelled" during my geotagging work i saw some code to fix issues however i don't no where i can write those one, i'm new here... someone can help me