mavlink/qgroundcontrol #6240: Save/load logic

AndKe, 2018-03-21:
To reproduce:
create a mission item.
click "sync" , "Remove all", confirm (at this point you expect to have no mission items..)
click "sync" , "Load from file" - gives "mission overwrite" dialog

Feedback:
-no reason to warn about "mission overwrite" when there is no mission.
-"remove all" should be disabled when there are no mission items anyway.
-I truly hate the mouse gymnastics, (click something on the left, then get an small ok/cancel dialog far to the right)  - this really stinks when using touchpad or "pointing stick"  - at least, there should be keyboard hotkeys for that (esc/enter ?)

DonLakeFlyer, 2018-03-21:
> at least, there should be keyboard hotkeys for that (esc/enter ?)

Yup. I've gone into hand to hand combat with QT on trying to make keyboard work multiple times. Failed every time. I guess I'll try again in 3.4.

DonLakeFlyer, 2018-03-21:
The trick to this one is you have to do it without a vehicle connected...

DonLakeFlyer, 2018-03-21:
> at this point you expect to have no mission items..

That ends up being a little tricky since this removes all items from the vehicle is connected. And it is possible for things to get out of sync, which would in turn prevent you from clearing the vehicle. So leaving as is. Since better to no-op then possibly lose functionality.

closed 2018-03-21 by DonLakeFlyer