mavlink/qgroundcontrol #6252: Feature request: add video stream from usb otg devices like app FPViewer for android

mkuliberda, 2018-03-23:
As in title it would be great to use qgroundcontrol together with available fpv usb UVC VIDEO DEVICES like this
5.8G Mini FPV Receiver UVC Video Downlink OTG VR Android Phone PC Micro USB APM Pix fishDrone GCS Collection Card Transmission
 http://s.aliexpress.com/aY3mua2Y?fromSns=Copy to Clipboard

taileron, 2018-08-16:
there will be probably a nice workaround. QGC displays extremely reliable super low latency rtsp streaming.
So it needs only a converter from usb cam to IP stream like "RTSP Cam Server", "IP Cam Pro" or "Periscope HD"  
and with an entry in this kind you get the stream in QGC:
rtsp://192.168.X.X:8554/live.sdp
Currently RCS does no usb and ICP does no RTSP yet - probably there is a further app that is capable of both on Android.

taileron, 2019-01-28:
"Nice Workaround" that´s a great converter: Vivotek VS8100-V2 works perfect, converts an analog video (from 5.8G FPV receiver) into a rtsp stream that can be displayed by QGC with similar latency to ROTG0X

mkuliberda, 2019-02-09:
Hi, thanks for all the replies and ideas. Unfortunately, I'd like to stay on the low cost, hobby side here and vivotek seems to be out of my prices range. I also couldn't find any android app that would work with aforementioned usb device and at the same time as a server to feed the stream to QGC. But for FPV with OSD I'm using at the moment FPViewer, and it works quite ok, maybe it will get streaming functionality in the future.