mavlink/qgroundcontrol #6255: terrain information not downloading

Antiheavy, 2018-03-24:
I'm having problems getting the terrain data to download or work on recent daily builds.  Tried several daily builds over the last week, most recent on 2018-03-23.  Tried this on several windows PCs and several different internet connections with similar results.

waypoints planned in a golf course with hills, no terrain shown:
[image]

"UnknownNetworkError" in console each time I place a waypoint:
[image]

am I doing something wrong?

DonLakeFlyer, 2018-03-25:
@dogmaphobic Any idea what QNetworkReply::UnknownNetworkError might mean? Todd was saying he isn't behind a proxy but he is behind a firewall.

DonLakeFlyer, 2018-03-25:
I've added better logging which hopefully will help figure this out. New build should be available in couple hours.

DonLakeFlyer, 2018-03-28:
@Antiheavy Have to tried very latest daily builds. It should hae additional TerrainQueryLog logging for these errors which will help me.

Antiheavy, 2018-03-28:
using 2018-03-27 daily build - this is what is says when I try to place waypoints.  same thing on 2 different PC's on 2 different internet connections (home and office).  see screenshot and also attached AppLog (zip file).
[image]
[terrain-not-working.zip](https://github.com/mavlink/qgroundcontrol/files/1857605/terrain-not-working.zip)

DonLakeFlyer, 2018-03-28:
Hmm, it's like it can't resolve the domain. It's not getting anything back.

DonLakeFlyer, 2018-03-28:
@dogmaphobic Do you have any ideas how to debug this?

DonLakeFlyer, 2018-04-01:
@NaterGator Nate could you see if terrain information downloads for you on Windows?

DonLakeFlyer, 2018-04-30:
This should be working now

closed 2018-04-30 by DonLakeFlyer