mavlink/qgroundcontrol #6256: "Upload Required" button flashes once then reverts to plain old "Upload" button.  Sync button does not flash.

Antiheavy, 2018-03-25:
In QGC Daily (2018-03-23) the "Upload Required" button only flashes once and then it changes back to the normal "Upload" button.  The "Sync" button also reverts to the normal look, not the flashing (!) look.  this seems like a bug?

[image]

DonLakeFlyer, 2018-03-25:
Is there something specific to landing pattern here? I can't get this to reproduce. Will need more detailed steps.

Antiheavy, 2018-03-25:
Here is a video of the Upload Required button and sync button auto-disappearing:
https://youtu.be/9zoTc0jsft4

DonLakeFlyer, 2018-04-29:
I think this may have been related to the various terrain download issues which have been fixed. Can you try to repro in new daily?

Antiheavy, 2018-04-30:
Yep! seems to be back to working in the new daily (tested on Windows 2018-04-29).  The Upload Required still disappears in our custom branch though, but I think the terrain fixes haven't been rolled into that custom branch properly so your hypothesis might make sense.  Anyway, closing this.

Antiheavy, 2018-04-30:
wait a second!  it's broke again when I switch to Feet measurement units (feet, acres, mph, F).  Both the altitude bars are broken and also the Upload Required button.... err....

Antiheavy, 2018-05-02:
okay, working in both feet and meters now with the latest Daily (maybe there was a scaling issue or something?).

closed 2018-05-02 by Antiheavy