mavlink/qgroundcontrol #6257: corridor scan stability/bugs

Antiheavy, 2018-03-25:
I'm seeing stability issues playing around with Corridor Scan in the 2018-03-23 Daily. 

When I draw or adjust a corridor scan I see a lot of these message spam in console:
[image]

also QGC crashes sometimes when creating or editing Corridor Scans (windows 10):
[image]

Antiheavy, 2018-03-25:
also, not sure if it is related, but the terrain data was also not working for me:
https://github.com/mavlink/qgroundcontrol/issues/6255

DonLakeFlyer, 2018-03-25:
> QGC crashes sometimes when creating

It crashes right after you click the Corridor Scan button?

DonLakeFlyer, 2018-03-25:
> editing Corridor Scans

Editing how. While you are dragging a point around or while you can changing a value?

DonLakeFlyer, 2018-03-25:
Found at least one of them. But sure if all yet. You can try new build in a bit and then if still crash need more specifics on repro steps.

Antiheavy, 2018-03-25:
> It crashes right after you click the Corridor Scan button?

sometimes, it is very intermittent.

> While you are dragging a point around

yes, sometimes, very intermittent.

> You can try new build in a bit and then if still crash

will do.

Antiheavy, 2018-03-25:
Still crashing on Daily 2018-03-25.  Application crash happens most often with clicking on the Corridor Scan button from the Pattern Toolbar.  

It seems to be 100% repeatable if I check the Console, see that I'm getting "NetworkError" messages related to TerrainQuery.  then go back to Plan screen, press the Corridor Scan button, ... crash.
Still happening on Daily 2018-03-25.

Antiheavy, 2018-03-25:
and if I wait until it starts showing ...."Downloading" 2 2" etc, it will still often crash when trying to create a Corridor Scan, but not always:
[image]

DonLakeFlyer, 2018-03-26:
The "bad vertext requested" log output are all gone now?

DonLakeFlyer, 2018-03-26:
I'd say it's a 75% chance I found and fixed it now!

Antiheavy, 2018-03-26:
> The "bad vertex requested" log output are all gone now?

i didn't notice them in the 2018-03-25 daily.  

> I'd say it's a 75% chance I found and fixed it now!

I'll try testing the next daily once it is out.

DonLakeFlyer, 2018-03-27:
New daily is available

Antiheavy, 2018-03-28:
I have not been able to repeat the crash using the 2018-03-27 daily build.  Seems you might have fixed it.

I still get "...UnknownNetworkError...." type message every time I place a waypoint or create a new vertex, but that must be a different problem.
[image]

closed 2018-03-28 by DonLakeFlyer