mavlink/qgroundcontrol #6262: 3.3 Joystick: Virtual RD by Joystick, arming impossible.

tubeme, 2018-03-25:
PX4 1.7.3
QGC 3.3
Joystick: Mad Catz F.L.Y.5
Flying Wing. (VTOL)

When the ARM button is not pressed first:

On arming with Joystick button, the flight mode automatically goes to Altitude and then I receive a warning that there is no 0 (zero) throttle and that is why it can arm. The throttle of the joystick stays at zero at that moment.

[image]

When the ARM button is pressed first:

On arming with joystick button the flight mode stays at Manual but again it announces the throttle is not zero and does not arm. After a few seconds it shows message Manual Control Lost.

[image]

DonLakeFlyer, 2018-03-26:
You have "Full down stick is zero throttle". Do you have throttle all the way down when you do these things?

james-mchugh, 2018-03-26:
I have been having trouble arming with Joystick as Virtual RC too, but when I try to arm, it says that there is no RC and fails.

tubeme, 2018-03-26:
@DonLakeFlyer Yes I have my throttle full way down at 0 all the time.. I can see that it is zero in the joystick settings.

DonLakeFlyer, 2018-03-30:
See #6264 for no RC problem.

I think arming problem may be caused by Circle Correction being disabled by default. I can't remember if you used to be able to arm using joystick. Was that possible in previous builds?

DonLakeFlyer, 2018-03-30:
Caused by this pull: https://github.com/mavlink/qgroundcontrol/pull/6177

DonLakeFlyer, 2018-03-30:
But the circle correction problem should only be in Daily not 3.3.

tubeme, 2018-03-31:
@DonLakeFlyer All the tests were made with the Stable branch and not the Daily. I think it is logical to be able to arm with the joystick. And now I program Arm and Disarm functions on two buttons on the joystick.

Before in the Virtual RC by Joystick  I was able to arm with full rudder commend like with the RC. But  this is disabled in 3.3.

closed 2024-06-02 by HTRamsey