mavlink/qgroundcontrol #6264: 3.3 Joystick: Joystick/ No RC Checks Manual Control lost.

tubeme, 2018-03-25:
PX4 1.7.3
QGC 3.3
Joystick: Mad Catz F.L.Y.5
Flying Wing. (VTOL)

When Joystick/ No RC Checks is used on arming it reports manual control lost and commands RTL which is rejected with a message Rejected RTL in a loop all the time until i press disarm.

[image]

DonLakeFlyer, 2018-03-30:
I believe this is caused by the fact that when you initiall boot QGC joystick commands don't automatically start sending. You have to go to the Joystick page to get them to start, even if joystick is enabled.

DonLakeFlyer, 2018-03-30:
@jaxxzer @nanthony21 Joystick folks. Could someone help out and fix this?

If you turn on JoystickValuesLog can you see that they don't start pumping out until you go to the joystick page. I believe this is the case with 3.3 as well so the fix needs to start on that branch.

nanthony21, 2018-03-30:
@DonLakeFlyer I dug into it a bit today. It looks like this was introduced by my pull request #6117. I'll fix it this weekend. It looks like the "Stable v3.3" branch is unaffected.

DonLakeFlyer, 2018-06-20:
This should be fixed now

closed 2018-06-20 by DonLakeFlyer