mavlink/qgroundcontrol #6266: Speech-to-text not working in latest stable and nightly in MacOS

anassinator, 2018-03-25:
This works perfectly fine with v3.2.7 on the same system (macOS 10.12.6). Wasn't able to test on Linux just yet.

Steps to reproduce:
1. Start QGC and make sure the "Mute all audio output" option is unticked
2. Connect to the telemetry kit
3. Send any `STATUSTEXT` message from the FCU with a high severity (for example by attempting to arm without pressing the enable switch)
4. Receive the `STATUSTEXT` message in a yellow popup, but notice that no voice is spoken.

Board: Pixhawk Mini
Firmware: PX4 v1.7.3

DonLakeFlyer, 2018-03-26:
@dogmaphobic Can you look?

dogmaphobic, 2018-03-26:
I could not reproduce this. TTS is working just fine here (master, Qt 5.9.3, macOS 10.13.3)

anassinator, 2018-03-26:
Are there some logs that I can produce that could help?

DonLakeFlyer, 2018-03-27:
Downloaded daily builds are not working for me either. @dogmaphobic Can you help with this? Likely builds are being built incorrectly somehow.

dogmaphobic, 2018-06-20:
@DonLakeFlyer The Qt blob (on S3) must be messed up. If I use it to run qmake, look at the last message:

```
~/build$ ./Qt5.11-clang_64/5.11.0/clang_64/bin/qmake -r ~/github/work/UpstreamQGC/qgroundcontrol.pro CONFIG+=installer -spec macx-clang
Info: creating stash file /Users/gus/build/.qmake.stash
Project MESSAGE: Qt version 5.11.0
Project MESSAGE: Mac build
Project MESSAGE: QGroundControl version 3.3.2 build 742 describe Development master:4dc7b9d64 2018-06-20 10:59:36 -0700
Project MESSAGE: Build Installer
Project MESSAGE: Release flavor
Project MESSAGE: Using Default QtLocation headers
Project MESSAGE: Using MAVLink dialect 'ardupilotmega'.
Project MESSAGE: Including support for Zeroconf (Bonjour)
Project MESSAGE: Including support for video streaming
Project MESSAGE: Preparing GStreamer Framework
Project MESSAGE: This project is using private headers and will therefore be tied to this specific Qt module build version.
Project MESSAGE: Running this project against other versions of the Qt modules may crash at any arbitrary point.
Project MESSAGE: This is not a bug, but a result of using Qt internals. You have been warned!
Empty filename passed to function
```

If I do the same using my local copy of Qt, it all works:

```
~/build$ ~/Applications/Qt/5.11.0/clang_64/bin/qmake -r ~/github/work/UpstreamQGC/qgroundcontrol.pro CONFIG+=installer -spec macx-clang
Project MESSAGE: Qt version 5.11.0
Project MESSAGE: Mac build
Project MESSAGE: QGroundControl version 3.3.2 build 742 describe Development master:4dc7b9d64 2018-06-20 10:59:36 -0700
Project MESSAGE: Build Installer
Project MESSAGE: Release flavor
Project MESSAGE: Using Default QtLocation headers
Project MESSAGE: Using MAVLink dialect 'ardupilotmega'.
Project MESSAGE: Including support for Zeroconf (Bonjour)
Project MESSAGE: Including support for video streaming
Project MESSAGE: Preparing GStreamer Framework
Project MESSAGE: This project is using private headers and will therefore be tied to this specific Qt module build version.
Project MESSAGE: Running this project against other versions of the Qt modules may crash at any arbitrary point.
Project MESSAGE: This is not a bug, but a result of using Qt internals. You have been warned!
```

Also, this logic is not clear to me. It seems all 3 tests are the same, causing it to download GStreamer and install it, AND download its blob and copy it. Is that intentional?

```
  - if [[ "${SPEC}" = "macx-clang" && "${TRAVIS_PULL_REQUEST}" = "false" ]]; then
        wget --quiet https://s3-us-west-2.amazonaws.com/qgroundcontrol/dependencies/gstreamer-1.0-1.5.2-x86_64.pkg &&
        sudo installer -verboseR -pkg gstreamer-1.0-1.5.2-x86_64.pkg -target /
        ;
    fi

  - if [[ "${SPEC}" = "macx-clang" && "${TRAVIS_PULL_REQUEST}" = "false" ]]; then
        wget --quiet https://s3-us-west-2.amazonaws.com/qgroundcontrol/dependencies/gstreamer-1.0-devel-1.5.2-x86_64.pkg &&
        sudo installer -verboseR -pkg gstreamer-1.0-devel-1.5.2-x86_64.pkg -target /
        ;
    fi

  - if [[ "${SPEC}" = "macx-clang" && "${TRAVIS_PULL_REQUEST}" = "false" ]]; then
        wget --quiet https://s3-us-west-2.amazonaws.com/qgroundcontrol/dependencies/osx-gstreamer.tar.bz2 &&
        sudo tar jxf osx-gstreamer.tar.bz2 -C /Library/Frameworks
        ;
    fi
```

DonLakeFlyer, 2018-06-20:
> Also, this logic is not clear to me. It seems all 3 tests are the same, causing it to download GStreamer and install it, AND download its blob and copy it. Is that intentional?

Don't know anything about that.

DonLakeFlyer, 2018-06-20:
Seems wrong. I don't know the state of the tar on S3. @dagar must have built that.

DonLakeFlyer, 2018-06-20:
> Empty filename passed to function

Any idea where that is coming from?

dogmaphobic, 2018-06-20:
> Any idea where that is coming from?

No, when I googled it, I got a bunch of ambiguous references but nothing that pointed straight to something. The closest I got was a case where there was a space in some path (either actual file or qrc). That doesn't make sense in this context.

DonLakeFlyer, 2018-06-20:
It's not happening on the travis builds though: https://travis-ci.org/mavlink/qgroundcontrol/jobs/394724307

DonLakeFlyer, 2018-06-20:
But then if you look at that log it also says ```Project MESSAGE: Skipping GStreamer Framework``` which is also wrong.

dogmaphobic, 2018-06-20:
That's probably a result of the weird logic above. When comparing the blob downloaded from S3 to my local copy, I noticed that all symlinks are actual files (duplicating their sizes) and many other files have different sizes even though they should be identical.

DonLakeFlyer, 2018-06-22:
Verified that CI built daily builds are now working.

closed 2018-06-22 by DonLakeFlyer