mavlink/qgroundcontrol #6269: Joystick inputs no longer visible unless calibrating

DonLakeFlyer, 2018-03-26:
You used to be able to move joysticks around and seem the results in the Joystick window. Now you can only see that when calibrating. This was broken by a recent change. Not sure if this is broken in 3.3 stable as well. I think the change that broke was related to the wierd tri-state is calibrating flag that was changed to a bool maybe.

@jaxxzer Did you make that state change? This is broken now.

DonLakeFlyer, 2018-03-28:
@jaxxzer Do you remember this change?

jaxxzer, 2018-03-29:
I don't think this was me.

I can try to help find the problem, though.

DonLakeFlyer, 2018-03-29:
Ok, thanks. Let me dig into the source a little to figure out who. Give me a day to do that before looking.

jaxxzer, 2018-03-29:
ok

DonLakeFlyer, 2018-03-29:
@nanthony21 Your change here: https://github.com/mavlink/qgroundcontrol/commit/02a8b5b1d0fc9551f319553bd4f36e7257fdfbf8#diff-7179de35ff534916a12fd2e757566fe8 has broken the ability to see joystick movement while you aren't calibrating. Could you fix that?

nanthony21, 2018-03-30:
@DonLakeFlyer I'm not seeing the issue you described on the current master branch. From what I see it always shows the joystick inputs on the right side of the page and the only time that it doesn't show the joystick values on the left side of the page is when joystick input is disabled. Do you have any advice on how to reproduce the issue?

DonLakeFlyer, 2018-03-30:
Let me try it again myself. Was getting a number of report from folks on this. That doesn't seem to match what I was seeing.

DonLakeFlyer, 2018-03-30:
Hmm, not sure what I was doing/seeing. Now I don't see it either. All working fun. @nanthony21 Sorry for the false alarm. Thanks for helping out.

closed 2018-03-30 by DonLakeFlyer