mavlink/qgroundcontrol #6281: Heavy system load with QGroundControl 3.3.0

estechnical, 2018-03-30:
This system is for posting bugs or feature requests ONLY. For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

----
When posting bug reports, include the following informaiton:
- Ground station operating system.
- QGroundControl version and build (daily, stable, self-built from source, etc.)
- Autopilot board: Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.
- Autopilot firmware (e.g. PX4, ArduPilot, custom) and version.
- Exact steps to reproduce the problem. Starting from booting QGroundControl to when the problem occurs.

**All this template text should be deleted before creating your issue.**
Running on Ubuntu 16.04
QGroundcontrol appimage 3.3.0 (also tested 3.3.1)
Pixhawk 2
PX4 firmware 1.7.3

I have QGroundControl making a tcp connection to mavlink-router which is connected to the telem2 port of px4. The Nuttx console (via mavlink) appears to lose short commands eg "ps" and I have to type them several times before I get the output. Also it seems very inconsistent compared to doing the same thing with the USB cable connected, top doesn't display well, like whole lines are lost (or just not redrawn).

I don't see any warnings/errors from mavlink-router.

Although CPU load isn't very high on the computer running qgroundcontrol, the system load goes very high, like it's io bound.

I notice top of list in `htop` is `/tmp/.mount_[random chars]/qgroundcontrol`

Is there logging that you can point me at as I'm interested to see why I'm getting this issue? Any other debugging options?

estechnical, 2018-03-30:
My initial suspicions seem incorrect, this happens too when connected with USB in v3.3.1 after a while of having things connected.

DonLakeFlyer, 2018-03-30:
> the system load goes very high

It seems like you are saying that disk io is high. Possibly paging. Are you running with small amounts of memory on the machine? Or is QGC leaking memory?

estechnical, 2018-03-30:
It doesn't look like it's leaking memory. But yes, heavy disk io or something io related is taking my system load up to 4.0 on a 4 cpu machine, while the actual CPU graphs usually stay lower than 50% across all cores. QGroundcontrol also seems to cause these messages to appear in dmesg, nothing else I use does this. It happens within a few minutes of starting.
```
[32220.120812] intel_powerclamp: Start idle injection to reduce power
[32256.260866] intel_powerclamp: Stop forced idle injection
[32276.282553] intel_powerclamp: Start idle injection to reduce power
[32308.349378] intel_powerclamp: Stop forced idle injection
```

My laptop is an i7 thinkpad with 16Gb ram, "only" 6~Gb used when I start Qgroundcontrol, climbs to about 6.5Gb, no signs of swap being used or even all the ram getting close to being allocated.

Ok, so I'm running chrome (15 tabs), sublime text (a bunch of windows each with tabs) and a terminal, but I'm not stressing the machine with this. Mavlink router doesn't seem to test the CPU much (4% reported by htop for the process when connected)

I'm a bit puzzled as it seems to cause 2 second freezes in input typing this message.

DonLakeFlyer, 2018-03-30:
https://askubuntu.com/questions/457252/intel-powerclamp-start-stop-forced-idle-injection

estechnical, 2018-04-01:
Thanks - after some digging I have arrived at the conclusion the fan speed control of my thinkpad was "not the same as before". 

I have installed `thinkfan` on my laptop and I'm testing this now. For anyone else encountering this issue, I followed these instructions, adapting for the sensors I discovered in my laptop. http://andrewjackson.name/blog/2016/08/07/set-up-thinkfan-on-lenovo-thinkpad-t430-ubuntu-16-04/

It appears like the default fan speed control never reaches top speed any more, thinkfan takes charge and seems to work well so far.

closed 2018-04-01 by DonLakeFlyer