mavlink/qgroundcontrol #6286: Resume Mission Not Working Fixed Wing

J1100, 2018-03-30:
I flew today with a fixed wing Ardupilot drone with daily build version 3.3.0-258-g0507fb9bd

The “Resume Mission” feature did not work as described. After initiating RTL, the drone came in to land. I simulated a battery change by unplugging the drone for about a minute and powering it up again. The drone reconnected to QGC ok, but there was no attempt to rebuild the mission and the only pop-up I received was “Start Mission,” when activated I just received an error that it wasn’t supported for my vehicle type. This pop-up came on screen many times during my flights even though it did nothing.
----
- Ground station operating system: Samsung Galaxy Tab E Android 7.1.1
- QGroundControl version and build: 3.3.0-258-g0507fb9bd
- Autopilot board: Pixhawk 
- Autopilot firmware: ArduPlane 3.7.1

[image]

DonLakeFlyer, 2018-03-31:
Hmm, the problem is that if you use a DO_LAND_START and RTL is set to go to there then the flight mode never changes from Auto to RTL. Which in turn breaks the ability the remember the last waypoint flown to. Does this work in Mission Planner?

J1100, 2018-03-31:
Yes, the flight mode did not change from Auto to RTL. I assumed this was QGC's work because when initiating RTL in fixed wing on Mission Planner, the drone comes and circles the home location at a predetermined altitude and does not land.

Yes, the drone does not remember the last WP it flew to. I assumed based on the description of Resume Mission in the QGC manual that QGC was remembering this waypoint, and using that information to "Rebuild" the mission. Now it sounds like nothing of the sort is happening.

In Mission Planner, it remembers the last flown to WP if taken out of Auto and flown in another mode, but if you interrupt the mission to Land, then it does not remember the last completed WP.

"Resume Mission" the way it is described is one of the main reasons why I am trying to use QGC as it will make the workload much easier for customers to use on large agricultural fields where they must change batteries.

DonLakeFlyer, 2018-03-31:
There are two problems here:
1) Start Mission is not supported by ArduPlane.
2) Resume Mission doesn't trigger if you have RTL_AUTOLAND != 0 and use RTL.

Workaround which may or may not be useful for now:
1) You can start the mission and takeoff manually. This will work with resume mission as well. USe the first slider which re-generates the mission. Then don't use the second slider which is essentially a Start Mission.
2) Resume Mission should trigger if RTL_AUTOLAND=0 and you RTL and then manually land the plane youself.

Talking with ArduPilot folks about how to fix/make these things work.

J1100, 2018-04-01:
"You can start the mission and takeoff manually. This will work with resume mission as well. USe the first slider which re-generates the mission. Then don't use the second slider which is essentially a Start Mission."

What "first slider" are you talking about? The one and only slider that pops up is "Start Mission"

Have you actually tested a fixed wing yourself with Resume Mission?

DonLakeFlyer, 2018-04-01:
When you exit out of Auto the waypoint will be remembered. You then land the plan manually. When the vehicle disarms you will be prompted with a Resume Mission slide to rebuild the mission from the last waypoint. If you slide this, the mission will be rebuilt. You will then get a second slider which tells you to review the generated mission and when ready confirm to restart. This second slider will not work since Start Mission is not working.

FYI: If you are using ARMING_REQUIRE=0 none of this will work correctly. That is another ArduPlane only problem which is being worked on.

J1100, 2018-04-02:
Ok, well that's not going to work. The drone is designed to be landed with the autopilot only. Consider it a feature request, then.

DonLakeFlyer, 2018-04-04:
ArduPilot is going to change this so that it makes Resume Mission work.

J1100, 2018-04-04:
That's great news! Thank you.

DonLakeFlyer, 2018-04-04:
Also Start Mission problem is already fixed in latest QGC daily builds.

DonLakeFlyer, 2018-04-30:
Everything from the QGC side of things is done now I believe. Work now remains on the ArduPilot side.

closed 2018-04-30 by DonLakeFlyer