mavlink/qgroundcontrol #6289: Repeating "not enough bandwidth for log streaming" messages

estechnical, 2018-03-31:
In QGC 3.3.1, I was experimenting with different baud rates for my connection. I set Normal telemetry 57600 for the telem 2 port and then got continuous messages being spoken about "not enough bandwidth for log streaming"

Whilst I understand the message and it's sensible to recommend having more bandwidth available, it's unusable with the computer continuously telling me about it.

dagar, 2018-03-31:
This is sent as a response to MAV_CMD_LOGGING_START - https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L548

QGC is trying to enable ulog streaming over and over again, but getting denied.

dagar, 2018-03-31:
@DonLakeFlyer could QGC stop once it's received a denied? https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L554

DonLakeFlyer, 2018-03-31:
Ah, I had no idea QGC was sending those commands. @dogmaphobic Can you look at this?

dogmaphobic, 2018-04-01:
I've added code in #6292 to deal with it. I could not test as nothing I have (working) denies logging.

closed 2018-04-30 by DonLakeFlyer

frank-qcd-qk, 2019-10-05:
Hi, I am using v3.5.4 (latest) and this issue still exists....Any workarounds?

DonLakeFlyer, 2019-10-05:
The fix is only available in daily builds

frank-qcd-qk, 2019-10-05:
> The fix is only available in daily builds

So can you point to me where I can get that daily builds and build my own?

frank-qcd-qk, 2019-10-05:
I also noticed that if I set to not emit heart beat that problem goes away..

DonLakeFlyer, 2019-10-05:
https://docs.qgroundcontrol.com/en/releases/daily_builds.html