mavlink/qgroundcontrol #6313: SurveyItemEditor assumes isotropic resolution

krokicki, 2018-04-06:
In 3.3, qgroundcontrol/src/PlanView/SurveyItemEditor.qml, the recalcFromCameraValues functions calculates ground resolution like this:

`groundResolution = (altitude * sensorWidth * 100) / (imageWidth * focalLength)`

The function ignores sensorHeight entirely, which would be fine if camera sensors had isotropic resolution, but my understanding is that they do not. At least it doesn't seem that way on my GoPro. 

Depending on whether portrait or landscape mode is used, this could miscalculate either side overlap or frontal overlap. 

If someone else can please confirm my reasoning above, I can submit a pull request to fix this by calculating separate x and y resolutions.

DonLakeFlyer, 2018-04-06:
Ok, looking at this a little more closely. The new code is exactly the same. I get what you are saying with respect to groundResolution now. But the issue is that it's not really clear how to calc that value when the camera is turned sideways on the vehicle. Since there is only a single ground resolution value, it's not super clear in what orientation that may be in.

@antiheavy Any ideas where?

Antiheavy, 2018-04-06:
> Don't most cameras have square pixels?

yes, most cameras (not all) have square pixels at the focal plane array (the CMOS or CCD sensor inside), but they of course get stretched, skewed, and distorted once projected onto the ground.  Factors include the type and quality of the lens, the orientation of the camera, and the contours of the ground.

GoPro cameras have high distortion lenses to achieve their very wide field of view.  This works awesome for emotionally charged action video shots, but not so great for ground mapping.  Some 3D reconstruction techniques actually take advantage of high distortion lenses so it's not all bad.  The cameras we manufacture use very low distortion lenses which correct out the typical lens distortion effects and optimize for ground mapping applications.

In my opinion, the potential variety of cameras and lenses out there are so great that all you can really do in QGC is make the simplest assumption which is to calculate GSD (ground sample distance, aka ground resolution) based on a square center pixel.

DonLakeFlyer, 2018-04-06:
> In my opinion, the potential variety of cameras and lenses out there are so great that all you can really do in QGC is make the simplest assumption which is to calculate GSD (ground sample distance, aka ground resolution) based on a square center pixel.

Which is essentially what the code does. So I think we are ok.

krokicki, 2018-04-06:
Fair enough. Thanks for considering!

closed 2018-04-06 by krokicki