mavlink/qgroundcontrol #6318: QGC daily build 04/05/2018 unable to display full user interface

r0gelion, 2018-04-06:
Trying to configure Intel Aero Ready to Fly Drone, using QGC daily build (e15f496ab) and got the next error:

[image]

Intel Aero drone is running PX4 firmware and here are more information about the software/firmware it uses:

BIOS_VERSION = Aero-01.00.16
OS_VERSION = Poky Aero (Intel Aero Linux Distro) 1.6.1 (pyro)"
AIRMAP_VERSION = 1.8
FPGA_VERSION = 0xc2
AeroFC firmware version = 1.7.4

QGC daily build (e15f496ab) in Windows 10.

I checked the stable version of QGC v3.3.1 and the full user interface was displayed and I was able to configure the drone, just got an error when the drone connected to QGC.

[image]

@DonLakeFlyer @dagar

DonLakeFlyer, 2018-04-06:
That's all firmware side and/or vehicle setup issues. I'd post there.

juvinski, 2018-04-06:
I had this error too, in the QT log windows there is several Error writing to QHostAddress("")

DonLakeFlyer, 2018-04-06:
Wait your daily is based off of a bad pull. First thing use a newer daily.

r0gelion, 2018-04-06:
> That's all firmware side and/or vehicle setup issues. I'd post there.

Thank you! I'll post this there.

DonLakeFlyer, 2018-04-06:
> Wait your daily is based off of a bad pull. First thing use a newer daily.

Try with newer daily build first to see if things work better. The specific one you are using has a bad bug in it.

juvinski, 2018-04-06:
I tried with last master - commit 189dc1cf115302215125537b1fdbf976e42f2ca8
and ardusim - with plane - and error still happening.

DonLakeFlyer, 2018-04-07:
Ok. Then not caused by that bad pull.

Antiheavy, 2018-04-15:
this error pops up for me all the time now in SITL using PX4 Master and recent QGC Dailies - seems like the UDP connection is dropping out or something, not sure:
[image]

DonLakeFlyer, 2018-04-30:
UDP problems are fixed now

closed 2018-04-30 by DonLakeFlyer