mavlink/qgroundcontrol #6329: Support PING protocol

mhkabir, 2018-04-08:
@DonLakeFlyer We need QGC support for returning the mavlink `PING` message (http://mavlink.org/messages/common#PING). I implemented the PX4 side here : https://github.com/PX4/Firmware/tree/pr-ping-protocol. It would be great if you could look into this. Thanks!

It needs QGC to return the ping messages (filled with sysid and compid of the system QGC received the ping from) so that we have a working test case.

See also : https://github.com/PX4/Firmware/issues/9233

DonLakeFlyer, 2018-04-10:
This doesn't make sense to me:
```
Unix timestamp in microseconds or since system boot if smaller than MAVLink epoch (1.1.2009)
```

What should time value be?

dogmaphobic, 2018-04-10:
Boot time is usually only for micro controllers. QGC would return `QDateTime::currentMSecsSinceEpoch() * 1000)`

DonLakeFlyer, 2018-04-10:
> if smaller than MAVLink epoch (1.1.2009)

This seems to indicate there is a different time format it wants.

mhkabir, 2018-04-10:
> This seems to indicate there is a different time format it wants.

The time should be in microseconds since UNIX epoch. We just use 1.1.2009 as a check for validity of the system's current RTC time.

DonLakeFlyer, 2018-04-16:
The mavlink spec needs a whole bucket of updating! So QGC is supposed to respond with a PING message which:
* Has the timestamp from the original message
* What does it do with the seq field?

mhkabir, 2018-04-16:
@DonLakeFlyer 
> Has the timestamp from the original message
Yes. (left untouched)
> What does it do with the seq field?
This is set from original message. (left untouched)

For the other two fields, sysid and compid : 
These will be set to the sysid and compid of the system which sent the original message to QGC.

mhkabir, 2018-04-29:
@DonLakeFlyer Any updates on this? Thanks!

DonLakeFlyer, 2018-04-29:
None. I just got back from being out of town for two weeks.

mhkabir, 2018-05-02:
This is done. Thanks a lot Don!

closed 2018-05-02 by mhkabir