mavlink/qgroundcontrol #6333: [Improvement] Calibration result texts of airspeed sensor disappear abruptly when finishing

bigbellmercy, 2018-04-10:
Problem:
While in airspeed calibration, text results of the airspeed calibration screen disappear abruptly right after blowing the sensor. So users cannot assure the calibration result succeeded or not.

Environment:
Ubutu 14.04, QGC v3.3.1

Thanks for excellent QGC!

DonLakeFlyer, 2018-04-10:
What type and version of firmware? Can you take a screenshot of the end result where you can't see the results any more.

bigbellmercy, 2018-04-10:
Firmware: v1.7.3 in Pixhawk2.1

Screen shot after finishing calibration of airspeed sensor:
[image]

Antiheavy, 2018-04-14:
> Problem:
> While in airspeed calibration, text results of the airspeed calibration screen disappear abruptly right after blowing the sensor. So users cannot assure the calibration result succeeded or not.

yes, this is annoying.  I don't know if it is a QGC thing or a firmware thing.  it seems like the text should persist as long as the view is open, even after completing the test, there should be a "success" type message left onscreen as the most recent message.   if the view is changed out of the Sensors tab, then it might still be nice if the calibration display zeroed out or whatver the term would be...

DonLakeFlyer, 2018-04-14:
My guess is the firmware started sending an airspeed cal finished text or something that wasn't there before. I believe the text used to stay up.

Antiheavy, 2018-04-14:
> My guess is the firmware started sending an airspeed cal finished text or something that wasn't there before. I believe the text used to stay up.

@dagar any insight here?

DonLakeFlyer, 2018-04-14:
I can fix from QGC side.

DonLakeFlyer, 2018-05-02:
> My guess is the firmware started sending an airspeed cal finished text or something that wasn't there before. I believe the text used to stay up.

Nope. They always disappeared. Level horizon has the same problem.

closed 2018-05-02 by DonLakeFlyer