mavlink/qgroundcontrol #6337: libssl and libcrypto not included in AppImage

patrickelectric, 2018-04-11:
last QGC AppImage release.
```
Cannot load library crypto: (crypto: cannot open shared object file: No such file or directory)
loaded library "crypto"
loaded library "ssl"
qt.network.ssl: QSslSocket: cannot resolve CRYPTO_num_locks
qt.network.ssl: QSslSocket: cannot resolve CRYPTO_set_id_callback
qt.network.ssl: QSslSocket: cannot resolve CRYPTO_set_locking_callback
qt.network.ssl: QSslSocket: cannot resolve ERR_free_strings
qt.network.ssl: QSslSocket: cannot resolve EVP_CIPHER_CTX_cleanup
qt.network.ssl: QSslSocket: cannot resolve EVP_CIPHER_CTX_init
qt.network.ssl: QSslSocket: cannot resolve sk_new_null
qt.network.ssl: QSslSocket: cannot resolve sk_push
qt.network.ssl: QSslSocket: cannot resolve sk_free
qt.network.ssl: QSslSocket: cannot resolve sk_num
qt.network.ssl: QSslSocket: cannot resolve sk_pop_free
qt.network.ssl: QSslSocket: cannot resolve sk_value
qt.network.ssl: QSslSocket: cannot resolve SSL_library_init
qt.network.ssl: QSslSocket: cannot resolve SSL_load_error_strings
qt.network.ssl: QSslSocket: cannot resolve SSL_get_ex_new_index
qt.network.ssl: QSslSocket: cannot resolve SSLv3_client_method
qt.network.ssl: QSslSocket: cannot resolve SSLv23_client_method
qt.network.ssl: QSslSocket: cannot resolve SSLv3_server_method
qt.network.ssl: QSslSocket: cannot resolve SSLv23_server_method
qt.network.ssl: QSslSocket: cannot resolve X509_STORE_CTX_get_chain
qt.network.ssl: QSslSocket: cannot resolve OPENSSL_add_all_algorithms_noconf
qt.network.ssl: QSslSocket: cannot resolve OPENSSL_add_all_algorithms_conf
qt.network.ssl: QSslSocket: cannot resolve SSLeay
qt.network.ssl: QSslSocket: cannot resolve SSLeay_version
qt.network.ssl: QSslSocket: cannot call unresolved function SSLeay
qt.network.ssl: Incompatible version of OpenSSL
qrc:/qml/APMSubFrameComponent.qml:178: TypeError: Cannot read property of null
qrc:/qml/APMSubFrameComponent.qml:179: TypeError: Cannot read property of null
qrc:/qml/APMSubFrameComponent.qml:180: TypeError: Cannot read property of null
qrc:/qml/APMSubFrameComponent.qml:189: TypeError: Cannot read property of null
qrc:/qml/APMSubFrameComponent.qml:190: TypeError: Cannot read property of null
qrc:/qml/APMSubFrameComponent.qml:191: TypeError: Cannot read property of null
file:///tmp/.mount_pSqfiM/Qt/qml/QtQuick/Controls/Private/Control.qml:88:5: QML Loader: Cannot anchor to an item that isn't a parent or sibling.
file:///tmp/.mount_pSqfiM/Qt/qml/QtQuick/Controls/Private/BasicButton.qml:189:5: QML MouseArea: Cannot anchor to an item that isn't a parent or sibling.
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
FirmwareUpgradeLog: PX4 releases github download failed "Error during download. Error: 99"
```

Same problem with firmware and parameter download.

DonLakeFlyer, 2018-04-11:
As far as I know these are expected to already be on the OS. Works fine on Ubuntu 16.04. From the errors it's not that they aren't there. They are just imcompatible with Qt.

patrickelectric, 2018-04-18:
This guy https://github.com/bluerobotics/ardusub/issues/147#issuecomment-382294275 have the same problem with debian. In my system I can see that I have /usb/lib/libcrypto.so.1, that's the weird thing. Running QGC with strace do not result in anything related to sll or crypto also. 
Besides that, adding libcrypto is not a solution, [AppImageKit Desktop-Linux-Platform-Issues](https://github.com/AppImage/AppImageKit/wiki/Desktop-Linux-Platform-Issues).

patrickelectric, 2018-04-18:
From [here](https://github.com/qt/qtbase/blob/50deb8cf70f61e21fb0c35182341477af11adbc1/src/network/ssl/qsslsocket_openssl_symbols.cpp#L745).
```
    // Try to find the libssl library on the system.
    //
    // Up until Qt 4.3, this only searched for the "ssl" library at version -1, that
    // is, libssl.so on most Unix systems.  However, the .so file isn't present in
    // user installations because it's considered a development file.
    //
    // The right thing to do is to load the library at the major version we know how
    // to work with: the SHLIB_VERSION_NUMBER version (macro defined in opensslv.h)
    //
    // However, OpenSSL is a well-known case of binary-compatibility breakage. To
    // avoid such problems, many system integrators and Linux distributions change
    // the soname of the binary, letting the full version number be the soname. So
    // we'll find libssl.so.0.9.7, libssl.so.0.9.8, etc. in the system. For that
    // reason, we will search a few common paths (see findAllLibSsl() above) in hopes
    // we find one that works.
    //
    // It is important, however, to try the canonical name and the unversioned name
    // without going through the loop. By not specifying a path, we let the system
    // dlopen(3) function determine it for us. This will include any DT_RUNPATH or
    // DT_RPATH tags on our library header as well as other system-specific search
    // paths. See the man page for dlopen(3) on your system for more information.
```

patrickelectric, 2018-04-26:
@DonLakeFlyer The problem appears to be with openssl 1.1 that does not have binary compatibility with 1.0;
It appears that Qt5.10 does not have this problem.
https://github.com/probonopd/linuxdeployqt/issues/209

patrickelectric, 2018-04-26:
I created a AppImage with 5.10.1 _**and**_ linuxdeployqt of QGC, it appears to be solved the problem, I'll do more tests with it.

DonLakeFlyer, 2018-04-29:
The AppImage we drop is only tested/supported on Ubuntu. Up until new that has mostly works on other Linux distros as well. Can we update docs to specify an install of libopenssl 1.0 prior to using QGC on those distros?

patrickelectric, 2018-04-30:
What we can do is to alarm the user about this issue with new versions of libopenssl, and ask to `export LD_LIBRARY_PATH=openssl-1.0_path_in_your_system`.

DonLakeFlyer, 2018-04-30:
I wonder if we just move latest master to 5.10 instead of trying to deal with this some other way.

patrickelectric, 2018-04-30:
I'm using it with 5.10 without problems.

jaxxzer, 2018-05-01:
I'll add for encouragement that when we get past 5.9.4, we will have the debug symbols for Qt.

DonLakeFlyer, 2018-05-01:
Wow, really? That's a big deal.

jaxxzer, 2018-05-01:
https://bugreports.qt.io/browse/QTBUG-3934

DonLakeFlyer, 2018-06-20:
Now on 5.11

closed 2018-06-20 by DonLakeFlyer

Williangalvani, 2018-09-27:
I'm running the 3.4.4 Appimage under Arch Linux:

logs:
(export QT_LOGGING_RULES="qt.network*.warning=true;qt.network*.debug=true")
When navigating to firmware upload section:
```
...
qt.network.ssl: QSslSocket: cannot resolve CRYPTO_num_locks
qt.network.ssl: QSslSocket: cannot resolve CRYPTO_set_id_callback
qt.network.ssl: QSslSocket: cannot resolve CRYPTO_set_locking_callback
qt.network.ssl: QSslSocket: cannot resolve ERR_free_strings
qt.network.ssl: QSslSocket: cannot resolve EVP_CIPHER_CTX_cleanup
qt.network.ssl: QSslSocket: cannot resolve EVP_CIPHER_CTX_init
qt.network.ssl: QSslSocket: cannot resolve sk_new_null
qt.network.ssl: QSslSocket: cannot resolve sk_push
qt.network.ssl: QSslSocket: cannot resolve sk_free
qt.network.ssl: QSslSocket: cannot resolve sk_num
qt.network.ssl: QSslSocket: cannot resolve sk_pop_free
qt.network.ssl: QSslSocket: cannot resolve sk_value
qt.network.ssl: QSslSocket: cannot resolve SSL_library_init
qt.network.ssl: QSslSocket: cannot resolve SSL_load_error_strings
qt.network.ssl: QSslSocket: cannot resolve SSL_get_ex_new_index
qt.network.ssl: QSslSocket: cannot resolve SSLv3_client_method
qt.network.ssl: QSslSocket: cannot resolve SSLv23_client_method
qt.network.ssl: QSslSocket: cannot resolve SSLv3_server_method
qt.network.ssl: QSslSocket: cannot resolve SSLv23_server_method
qt.network.ssl: QSslSocket: cannot resolve X509_STORE_CTX_get_chain
qt.network.ssl: QSslSocket: cannot resolve OPENSSL_add_all_algorithms_noconf
qt.network.ssl: QSslSocket: cannot resolve OPENSSL_add_all_algorithms_conf
qt.network.ssl: QSslSocket: cannot resolve SSLeay
qt.network.ssl: Incompatible version of OpenSSL
Map Cache in: "/home/will/.cache/QGCMapCache300" / "qgcMapCache.db"
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_CTX_new
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function ERR_get_error
qt.network.ssl: QSslSocket: cannot call unresolved function ERR_get_error
Temp log "/tmp/FlightData084559.mavlink"
libpng warning: iCCP: known incorrect sRGB profile
FirmwareUpgradeLog: PX4FirmwareUpgradeThreadController::findBoard
FirmwareUpgradeLog: 0 "PX4 FMU V2"
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_CTX_new
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function ERR_get_error
qt.network.ssl: QSslSocket: cannot call unresolved function ERR_get_error
FirmwareUpgradeLog: PX4 releases github download failed "Error during download. Error: 99"
```

And trying to load default parameters on ArduSub:
[image]

ldd output:
```
$ ldd ./QGroundControl3.4.4.appimage
	linux-vdso.so.1 (0x00007ffec39f4000)
	libfuse.so.2 => /usr/lib/libfuse.so.2 (0x00007fc50778c000)
	libpthread.so.0 => /usr/lib/libpthread.so.0 (0x00007fc50776b000)
	libglib-2.0.so.0 => /usr/lib/libglib-2.0.so.0 (0x00007fc507649000)
	libz.so.1 => /usr/lib/libz.so.1 (0x00007fc507432000)
	libc.so.6 => /usr/lib/libc.so.6 (0x00007fc50726e000)
	libdl.so.2 => /usr/lib/libdl.so.2 (0x00007fc507269000)
	/lib64/ld-linux-x86-64.so.2 => /usr/lib64/ld-linux-x86-64.so.2 (0x00007fc507a58000)
	libpcre.so.1 => /usr/lib/libpcre.so.1 (0x00007fc506ff5000)

```
My Openssl version(s):
```
core/openssl 1.1.1-1 [installed]
    The Open Source toolkit for Secure Sockets Layer and Transport Layer Security
core/openssl-1.0 1.0.2.p-1 [installed]
    The Open Source toolkit for Secure Sockets Layer and Transport Layer Security
```

@patrickelectric replicated the issue.

@DonLakeFlyer Can we re-open this?

DonLakeFlyer, 2018-09-27:
Some Linux folks are going to have to figure out how to fix this without breaking Ubuntu which is working fine. I don't really know anything about AppImage or how it works. I'm not sure the answer is to include the libs in the AppImage.

@thomasgubler I seem to remember from long, long, long ago that maybe you were an archlinux user. Any ideas?

patrickelectric, 2018-09-28:
@Williangalvani can you confirm if https://github.com/mavlink/qgroundcontrol/issues/6337#issuecomment-385387313 works ?

Williangalvani, 2018-09-28:
> @Williangalvani can you confirm if [#6337 (comment)](https://github.com/mavlink/qgroundcontrol/issues/6337#issuecomment-385387313) works ?
> Should we include openssl-1.0 ?

Yes, it works fine with `export LD_LIBRARY_PATH=/usr/lib/openssl-1.0`

kikislater, 2018-09-28:
Even with building from AUR repository with 3.4.4, it fails with libssl 1.1 at this time. Was previously working !

kikislater, 2018-09-28:
@Williangalvani : export works fine on archlinux. Thank you for it

DonLakeFlyer, 2018-09-28:
@patrickelectric This would seem to be a Qt bug wouldn't it?

patrickelectric, 2018-10-02:
> @patrickelectric This would seem to be a Qt bug wouldn't it?

this is related to the openssl library, when you compile and configure Qt, there is a "-DOPENSSL_API_COMPAT" parameter to select the compatibility with different versions

akontsevich, 2019-10-17:
Is there any working solution for this? Some URL to see the example? As I still get the error if build Qt and QGC in CentOS 7. I'm building AppImage with `appimagetool-x86_64.AppImage` - replaced ancient `AppImageAssistant` still used in QGC here: https://github.com/mavlink/qgroundcontrol/blob/master/deploy/create_linux_appimage.sh. Whether `linuxdeployqt` solves this problem? Thanks!